LK8000/LK8000 #576: Focus on window not working in config correctly

pventafridda, 2015-02-08:
Linux version. Simulate. Directly go to System setup, and for example enter page13, try to change Screen Data. When you click to change to half overlays you will not get the choice selected and be back to list of choices . If you click on Help instead of Half overlays, then you will be able to select half overlays, the focus been working.

brunotl, 2015-02-09:
focused & selected are two different concept !

List Item highlight are focus, so only painted when parent (list) have focus and of course, selection is not visible if list don't have focus.
 - on enter : focus are on Select button (first child control )
 - after close helps : focus are on Help Button ( focused by click )

we need to add "Selected item highlight" concept, for avoid this kinds of mistake.

I'll do that to Night.

pventafridda, 2015-02-09:
Thanks, that’s what I meant. The Selected item is already in use, it just does not work in such case for the dialogs when started up. Sort of..

From: Bruno de Lacheisserie 
Sent: Monday, February 09, 2015 1:17 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Focus on window not working in config correctly (#576)

focused & selected are two different concept !

List Item highlight are focus, so only painted when parent (list) have focus and of course, selection is not visible if list don't have focus.

  a.. on enter : focus are on Select button (first child control ) 
  b.. after close helps : focus are on Help Button ( focused by click ) 
we need to add "Selected item highlight" concept, for avoid this kinds of mistake.

I'll do that to Night.

brunotl, 2015-02-21:
fixed

closed 2015-02-24 by pventafridda