LK8000/LK8000 #596: Sound error ALSA lib

pventafridda, 2015-02-25:
during runtime we often get  ALSI lib pcm.c 7339 snd_pcm_recover  underrun occured.

brunotl, 2015-02-25:
problem is known, is in AlsaLib, install "pulseaudio" package solve the problem.

pventafridda, 2015-02-25:
Is it related to the PC version also? Because on PC there are problems with sounds on startup, there is a lock or hang pending..

brunotl, 2015-02-25:
Alsa on PC ?
- on Windows platform, we use win32 "sndPlaySound" API 
https://msdn.microsoft.com/en-us/library/windows/desktop/dd798676%28v=vs.85%29.aspx

- on Linux we Use SDL_Mixer 
http://www.libsdl.org/projects/SDL_mixer/docs/SDL_mixer.html

pventafridda, 2015-02-25:
For this reason I asked. Because we have a problem on PC, on startup the sound is muted, and the clockwork is running, and I have seen some crashes.

From: Bruno de Lacheisserie 
Sent: Wednesday, February 25, 2015 7:04 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Sound error ALSA lib (#596)

Alsa on PC ?

  a.. on Windows platform, we use win32 "sndPlaySound" API 
  https://msdn.microsoft.com/en-us/library/windows/desktop/dd798676%28v=vs.85%29.aspx

  b.. on Linux we Use SDL_Mixer 
  http://www.libsdl.org/projects/SDL_mixer/docs/SDL_mixer.html

pventafridda, 2015-02-25:
pulseaudio was installed on linux nimbus, the ALSA lib underrun error is always occurring.

pventafridda, 2015-03-19:
it disappeared automagically. mistery

closed 2015-03-19 by pventafridda