LK8000/LK8000 #598: Terrain Shading artefact

brunotl, 2015-02-25:
- Terrain Slope calculation are wrong on right border of screen.
- also wrong on border on top/right border of terrain data (DEM file)

PS : more or less visible, depends of Screen resolution, huge on 300*400.

pventafridda, 2015-02-25:
I cannot confirm this. Testing on Linux 400x300, demo terrain, no problems.

From: Bruno de Lacheisserie 
Sent: Wednesday, February 25, 2015 11:05 AM
To: LK8000/LK8000 
Subject: [LK8000] Terrain Shading artefact (#598)

  a.. Terrain Slope calculation are wrong on right border of screen. 
  b.. also wrong on border on top/right border of terrain data (DEM file) 
PS : more or less visible, depends of Screen resolution, huge on 300*400.

brunotl, 2015-02-25:
400x300 is not same as 300x400

| V5.0 with Shading     | V5.0 without Shading     |
| ------------- |:-------------:|
| [image]      | [image]             |

pventafridda, 2015-02-25:
Ok now I can see it in my linux too. But with light terrain it is quite hard to see it

From: Bruno de Lacheisserie 
Sent: Wednesday, February 25, 2015 9:58 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Terrain Shading artefact (#598)

400x300 is not same as 300x400

      V5.0 with Shading V5.0 without Shading

commit 97c5c094 (2015-02-27): Fix #598 TerrainDraw : remove wrong & useless rect visible

- if is biggers to Height buffer  -> test are always true so useless cpu overhead
( arround 4*height*with/dtquant integer compare, more than 260000 for 480*272 screen )
-if is smaller -> drawing artefact on border of terrain.

if one days someone try to enable RASTER_CACHE, don't forget to fix that.
closed 2015-02-27 by pventafridda