LK8000/LK8000 #600: CpuLoad don't work on linux.

brunotl, 2015-02-25:
we can see value upper than 130% on kobo

"getloadavg" report number of processes in the system run queue. it's not cpu load.
cf man page: http://linux.die.net/man/3/getloadavg

pventafridda, 2015-02-25:
I know perfectly. It is intentional and for me it is correct.
Load average is about system load, not process load. If the system is hung because of another process, it does not matter if LK is using only 10% if the system is overloaded.
We must show system load, and this is the most effective and simple way of doing it, at system level. 130% means system is busy for sure.
Run queue is not total number of processes. See on linux “uptime” .

From: Bruno de Lacheisserie 
Sent: Wednesday, February 25, 2015 10:17 PM
To: LK8000/LK8000 
Subject: [LK8000] CpuLoad don't work on linux. (#600)

we can see value upper than 130% on kobo

"getloadavg" report number of processes in the system run queue. it's not cpu load.
cf man page: http://linux.die.net/man/3/getloadavg

pventafridda, 2015-02-25:
If 130 is wrong, then the number of CPU cores in kobo is more than 1, but HaveSystemInfo did not detected and assumed only 1.
See function CpuSummary()

From: Bruno de Lacheisserie 
Sent: Wednesday, February 25, 2015 10:17 PM
To: LK8000/LK8000 
Subject: [LK8000] CpuLoad don't work on linux. (#600)

we can see value upper than 130% on kobo

"getloadavg" report number of processes in the system run queue. it's not cpu load.
cf man page: http://linux.die.net/man/3/getloadavg

brunotl, 2015-02-25:
Kobo cpu is : ARM Cortex A8 800 MHz single-core Freescale MCIMX507CVM8B

commit 61197418 (2015-02-26): Cpu load : use real cpu usage instead of averaged number of processes in the system run queue

previous method can be enabled using "USE_LOADAVG_CPU" define in option.h

fix #600
closed 2015-02-26 by pventafridda