LK8000/LK8000 #606: Fonts linux-pc problem..

pventafridda, 2015-02-26:
Example  480x272  dlgStartup.cpp using LK8MapFont , which is 
propGetFontSettingsFromString(TEXT("20,0,0,0,400,0,0,0,0,0,0,3,2,Tahoma"), &logfontMap);

but on Linux the vertical size (height) of the font generated is not the same as on PC.
Font is bigger in fact on linux.

brunotl, 2015-02-26:
font is Tahoma on PC and PNA, not on linux ...
for know what font is used on linux, look "Common/Source/xcs/Screen/Custom/Files.cpp"

pventafridda, 2015-02-26:
I know very well that Tahoma fonts belong to Microsoft and are not used in Linux.
What I am saying is that it happens that the font we are using are not matching the desired size.
If we ask for a font size 20 , like in the example, we expect 20 not 21.

From: Bruno de Lacheisserie 
Sent: Thursday, February 26, 2015 7:55 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Fonts linux-pc problem.. (#606)

font is Tahoma on PC and PNA, not on linux ...
for know what font is used on linux, look "Common/Source/xcs/Screen/Custom/Files.cpp"

pventafridda, 2015-04-15:
I am wrong. 
You are right. Closing.

closed 2015-04-15 by pventafridda