LK8000/LK8000 #610: Kobo: change Empty Map color to White as default.

SergioDaSilva82, 2015-02-27:
Having white as default looks much better...

brunotl, 2015-02-27:
only for B&W dithered device

SergioDaSilva82, 2015-02-27:
That is why I put Kobo in the title :)

pventafridda, 2015-02-27:
What if we create a device, like for Mio, Hp etc.?
Either we have embedded ifdef KOBO or we do like that.
Adding a device is much easier. All peculiar things belonging to kobo will be set that way.
I think that setting white background is not a problem for users, with or without kobo.

From: Bruno de Lacheisserie 
Sent: Friday, February 27, 2015 9:45 PM
To: LK8000/LK8000 
Subject: Re: [LK8000] Kobo: change Empty Map color to White as default. (#610)

only for B&W dithered device

SergioDaSilva82, 2015-02-27:
I agree that white for everyone won't be a problem. The fact is the LCD Green is not good for kobos.

pventafridda, 2015-02-28:
I confirm KOBO is a define., so we can put it at once

closed 2015-02-28 by pventafridda

brunotl, 2015-02-28:
KOBO device is not the right way
Use isDithered() instead

pventafridda, 2015-02-28:
seen late. good idea thanks i correct it