LK8000/LK8000 #611: Add menu for Kobo only

jaaaaf, 2015-02-27:
We need to have a new menu to set kobo functionalities.
On the first page with Fly, Profile, Simulate and Exit button, we can add "Kobo" to launch normal kobo reader (nickel) and a "system" menu to launch a new page for setting kobo system relative functionalities.
The new system page can have 5 buttons: Wifi, ftp, telnet, USB and close. First 4 buttons set on and off relative services/functionalities.

I can do it if integrated directly in same binary code, just adding buttons, system page and functionalities. If you think it's better to organize it differently, launching a first binary with the system page functionalities, who can launch the "regular" LK8000 binary, I will need some help to have an empty template displaying a page.

pventafridda, 2015-02-27:
no-
kobo users need a launcher, which is a separate thing from LK.

From: jaaaaf 
Sent: Friday, February 27, 2015 10:23 PM
To: LK8000/LK8000 
Subject: [LK8000] Add menu for Kobo only (#611)

We need to have a new menu to set kobo functionalities.
On the first page with Fly, Profile, Simulate and Exit button, we can add "Kobo" to launch normal kobo reader (nickel) and a "system" menu to launch a new page for setting kobo system relative functionalities.
The new system page can have 5 buttons: Wifi, ftp, telnet, USB and close. First 4 buttons set on and off relative services/functionalities.

I can do it if integrated directly in same binary code, just adding buttons, system page and functionalities. If you think it's better to organize it differently, launching a first binary with the system page functionalities, who can launch the "regular" LK8000 binary, I will need some help to have an empty template displaying a page.

SergioDaSilva82, 2015-02-27:
Maybe something like this: http://www.mobileread.com/forums/showpost.php?p=2595985&postcount=2

pventafridda, 2015-02-27:
I am closing this request. It doesnt belong to LK, strictly. Maybe we can open another project for it. The shell launcher for kobo is just another matter.

closed 2015-02-27 by pventafridda

brunotl, 2015-02-28:
It's joke ?

jaaaaf, 2015-02-28:
Like Bruno, I don't really understand you Paolo.
Actually we can use the Kobo because Bruno allow USB connection at startup
of LK8000. Without USB connection, we can't transfer file.
FTP or telnet will be actually useful to debug some problem.
Wifi can be useful in future to use some functionalities link to this
service
And if we want that pilots use LK8000 on Kobo, we can't say to them: If you
use LK8000, you can't use your kobo anymore to read book!

pventafridda, 2015-02-28:
I dont understand what you are talking about. 
This is unrelated to LK8000.
You are talking of doing a revolution on the startup pages to perform what instead should be done by a launcher.
If you dont understand this, why discussing further?
From: jaaaaf 
Sent: Saturday, February 28, 2015 4:40 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Add menu for Kobo only (#611)

Like Bruno, I don't really understand you Paolo.
Actually we can use the Kobo because Bruno allow USB connection at startup
of LK8000. Without USB connection, we can't transfer file.
FTP or telnet will be actually useful to debug some problem.
Wifi can be useful in future to use some functionalities link to this
service
And if we want that pilots use LK8000 on Kobo, we can't say to them: If you
use LK8000, you can't use your kobo anymore to read book!

jaaaaf, 2015-02-28:
Ok, no need to discuss further.

pventafridda, 2015-02-28:
In order to be perfectly clear, I try to explain myself without any possible interpretation.
What Kobo may need is a launcher, which is a program with all the buttons and eventually submenus that you mentioned in the first post, to perform system operations. From that program you run LK, as well.

jaaaaf, 2015-02-28:
I prefer this interpretation!

pventafridda, 2015-02-28:
Thats what I have been saying since the beginning! It should be quite fun to do it. But I am quite busy with 600x800 right now

pventafridda, 2015-02-28:
You should open an issue like "Create a launcher for Kobo"..
or maybe there is already something for the purpose.