LK8000/LK8000 #612: 10Hz variometer bar refresh rate & 1Hz screen refresh rate (LK8000 6.0?)

SergioDaSilva82, 2015-02-27:
Some pilots are requesting higher refresh rate for variometer bar 10Hz.

http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=7702

brunotl, 2015-11-16:
when we fly, we look outside, so it's useless ...

pventafridda, 2015-11-16:
that was the idea, in fact! (to look outside)

From: Bruno de Lacheisserie 
Sent: Monday, November 16, 2015 11:19 PM
To: LK8000/LK8000 
Subject: Re: [LK8000] 10Hz variometer bar refresh rate & 1Hz screen refresh rate (LK8000 6.0?) (#612)

when we fly, we look outside, so it's useless ...

SergioDaSilva82, 2015-11-16:
Paolo, so is better to close it.

closed 2015-11-17 by pventafridda

lshachar, 2018-08-14:
The on screen variometer is the only place where I get the actual vario value when I'm flying. I have a no screen vario and lk8000 is the only display I'm using. to me, this is more than useful, it is critical to have a much faster vario display. (personally I'm not using the bar, but only the numerical vario value)
(I got here from [this thread](http://www.postfrontal.com/forum/topic.asp?TOPIC_ID=7702) in postfrontal)

villamany, 2018-09-29:
It is important for me also, i would like to have a more responsive visual vario display (both vario bar and vario value overlay)