LK8000/LK8000 #613: Kobo: Infopages 2.x and 3.x need improvements

SergioDaSilva82, 2015-02-28:
Columns are too close, is hard to read the information.

Tested with Demo maps, demo airspaces and demo waypoints.

pventafridda, 2015-02-28:
Fixing in the weekend

From: SergioDaSilva82 
Sent: Saturday, February 28, 2015 4:14 AM
To: LK8000/LK8000 
Subject: [LK8000] Kobo: Infopages 2.x and 3.x need improvements (#613)

Columns are too close, is hard to read the information.

Tested with Demo maps, demo airspaces and demo waypoints.

SergioDaSilva82, 2015-02-28:
Do you think 1.x info pages also need some adjustments?

pventafridda, 2015-02-28:
no
---
Paolo Ventafridda

pventafridda, 2015-02-28:
OK the problem is that in the west world we write text horizontally and paint it like that.
With a portrait mode screen, there is no space for all informations if we enlarge the fonts.
I can change the way LK manage informations in portrait mode, but it is not a fast change and I am not sure we have to do it in 5.4.
I think about it today

From: SergioDaSilva82 
Sent: Saturday, February 28, 2015 4:14 AM
To: LK8000/LK8000 
Subject: [LK8000] Kobo: Infopages 2.x and 3.x need improvements (#613)

Columns are too close, is hard to read the information.

Tested with Demo maps, demo airspaces and demo waypoints.

SergioDaSilva82, 2015-02-28:
what do you think to have something like Android cellphones, the option to have small, normal or large fonts?

pventafridda, 2015-02-28:
I am not interested to spend weeks or months to do a revolution on fonts, sergio.

From: SergioDaSilva82 
Sent: Saturday, February 28, 2015 5:55 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Kobo Mini: Infopages 2.x and 3.x need improvements (#613)

what do you think to have something like Android cellphones, the option to have small, normal or large fonts?

pventafridda, 2015-04-19:
Now that fonts are rendered correctly I can rework the portrait mode to make it better.
It is in my queue officially. This is going to be available for kobo first.

closed 2015-04-27 by pventafridda