LK8000/LK8000 #619: Logbook: bitmaps error

pventafridda, 2015-03-06:
The scrolling cursor in logbook has disappeared, in both pc and linux

pventafridda, 2015-03-09:
Bitmaps are not initialized correctly all the times. It looks like something wrong is going on randomly.

SergioDaSilva82, 2015-03-22:
I had the same issue, I notice that happened to me when I copy the whole LK8000 folder from LINUX to the PC. but if I just copy the .exe file and the language folder over the LK8000 5.0a folder it works properly.

pventafridda, 2015-03-22:
Ah! You mean I am experiencing a config error!! I am going to check then (and also the logs.. )
thanks

From: SergioDaSilva82 
Sent: Sunday, March 22, 2015 2:31 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Logbook: bitmaps error (#619)

I had the same issue, I notice that happened to me when I copy the whole LK8000 folder from LINUX to the PC. but if I just copy the .exe file and the language folder over the LK8000 5.0a folder it works properly.

pventafridda, 2015-03-22:
After quick check for me it does not work

SergioDaSilva82, 2015-03-22:
just give a try about the way I did and check if your issue is fixed.

SergioDaSilva82, 2015-03-22:
can you send a screen shot about how it looks in your?

I had to make a clean install.

pventafridda, 2015-03-22:
it is the scrollbar bitmaps generally not painted on PC, on any page. 
.

brunotl, 2015-03-22:
You have wrong bitmap

SergioDaSilva82, 2015-03-22:
Bruno,

remember I had the same issue? I think I overwrite the new _Bitmaps folder
with the one from 5.0A release

brunotl, 2015-03-22:
Bitmap has changed if you use bitmap from 5.0 you have problem for sûre

pventafridda, 2015-03-22:
How can I have wrong bitmaps on PC? They are still the old SCROLLBARTOP etc. BMP .
Nothing changed.

From: Bruno de Lacheisserie 
Sent: Sunday, March 22, 2015 6:39 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Logbook: bitmaps error (#619)

You have wrong bitmap

pventafridda, 2015-03-22:
why did they change, we are still using GDI on Windows..
they were 32x32 4 bit in that case.
Now they are 64x32 8 bit .

From: Bruno de Lacheisserie 
Sent: Sunday, March 22, 2015 6:43 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Logbook: bitmaps error (#619)

Bitmap has changed if you use bitmap from 5.0 you have problem for sûre

SergioDaSilva82, 2015-03-22:
Paolo, try to use theses, were changed 5 months ago:

https://github.com/brunotl/LK8000/blob/master/Common/Distribution/LK8000/_System/_Bitmaps/SCROLLBARTOP.BMP

https://raw.githubusercontent.com/brunotl/LK8000/master/Common/Distribution/LK8000/_System/_Bitmaps/SCROLLBARBOT.BMP

https://github.com/brunotl/LK8000/blob/master/Common/Distribution/LK8000/_System/_Bitmaps/SCROLLBARMID.BMP

brunotl, 2015-03-22:
5.0 bitmap are wrong ! But if you prefer to don't trust me it's your choice

pventafridda, 2015-03-22:
Oh now I see why.. 

From: Bruno de Lacheisserie 
Sent: Sunday, March 22, 2015 6:43 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Logbook: bitmaps error (#619)

Bitmap has changed if you use bitmap from 5.0 you have problem for sûre

closed 2015-03-22 by pventafridda