LK8000/LK8000 #627: Canvas DrawText  crashes

pventafridda, 2015-03-19:
Passing unusual text, this function is crashing

brunotl, 2015-03-19:
in debug mode, if text contains invalid utf8 characters, assert is triggered.

the bug is not in DrawText, is in text source, invalid text can in very limited case :
  - invalid cast 
  - memory overflow or unterminated string
  - unchecked user input 

in all case bug need to be fixed at the source ...

pventafridda, 2015-03-19:
I know , I have found the problem. I remove the issue.

From: Bruno de Lacheisserie 
Sent: Thursday, March 19, 2015 4:57 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Canvas DrawText crashes (#627)

in debug mode, if text contains invalid utf8 characters, assert is triggered.

the bug is not in DrawText, is in text source, invalid text can in very limited case :

  a.. invalid cast 
  b.. memory overflow or unterminated string 
  c.. unchecked user input 
in all case bug need to be fixed at the source ...

closed 2015-03-19 by pventafridda