LK8000/LK8000 #630: MapWindow thread still getting input while on dialogs

pventafridda, 2015-03-23:
From main map, menu config Basic settings, for example.
Click on the blue background of the settings' window, near wind load , between Wing load and QNH or in the space between efficiency and Altitude. You hear the multimap sound and map is changing in the background.
I am not sure if this may lead to critical crashes or not.

brunotl, 2015-03-23:
not real problem, but fixed.

pventafridda, 2015-03-23:
great thanks
i think it could instead be a critical problem in flight.

From: Bruno de Lacheisserie 
Sent: Monday, March 23, 2015 3:24 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] MapWindow thread still getting input while on dialogs (#630)

not real problem, but fixed.

commit 49c64fcb (2015-03-26): fix #630, mouse event on  window derived from WindowControl need  to  always return true, otherwise event is forwarded to Parent Window.
closed 2015-03-26 by pventafridda