LK8000/LK8000 #631: LKASSERT Library/PressureSensor.cpp line 38

pventafridda, 2015-03-23:
While using Basic Settings experimenting another bug, got this message while exiting basic settings, I think. 
LK crashed without me doing anything, after a few seconds.
The assert line is about gliderpolar::bestld  found to be zero, which is not good.

pventafridda, 2015-03-23:
Probably the problem is in SetBallast in McReady.cpp . bestld is set to 0 there while calculating, under LockFlightData condition. SetBallast is run by the dlgBasicSettings.
In McReady there are a lot of LKASSERT checking bestld, except on initialization when we deliberately set it to 0, which is probably the problem. 
We must replicate it by entering and exiting ballast change in basic settings, until the problem appears. Once replicated on master branch, we must check for v5 as well because probably the problem is also there.
A possible quick and dirty solution for v5 is in AirDensitySinkRate setting bestld to 1.
Or even better, not using bestld 0 is McReady.

SergioDaSilva82, 2015-03-23:
Paolo, I got this picture from a pilot using Kobo:
[image]

pventafridda, 2015-03-23:
Very interesting! The AirDesntity rho is set to zero?!?
Only possible reason is a huge anomaly in the altitude.
The "pow" function can set result to zero if an overflow occurs, among other possible reasons. The only variable in the function is the altitude, which is also checked for overflow (positive only) greater than 44330m.
Since it is a signed double, I can only think of a negative altitude incoming (which is the equivalent of a huge positive altitude exceeding the sing bit). 
This is however a different issue not related to the other one mentioned in this thread.
It is a managed exception, exposed only in test versions to show the possible problem.
In a stable version nothing would have happened, except that the user would never know that his instrument is passing wrong values.
So first thing is to ask the user what is he using (baro altitude? incoming from which instrument? gps? from which source?).

pventafridda, 2015-03-23:
I cant stress enough the use of LKASSERT in place of assert(). We do nothing with assert, they are not useful for real usage outside our own computers, unless they appear on the screen and log themselves somewhere, which is why we have LKASSERT.

pventafridda, 2015-03-23:
For SetBallast() problem, a more likely reason is that SinkRate() is buggy and returned a positive value. In such case ld at line 97 is not bigger than zero, and bestld is never changed from the initial zero. Then we unlock FlightData and the crash happens.

I dont remember if we have changed the SinkRate in the master, I remember some work in McReady not too much time ago.

SergioDaSilva82, 2015-03-23:
Regarding the error on line123, I got the info that was using BlueFlyVario with 10Hz.

Inside of house.

pventafridda, 2015-03-23:
try to replicate it while doing an NMEA log. At 10hz I have never checked anything so far.

SergioDaSilva82, 2015-03-23:
Is it possible to make LK8000 create NMEA automatically in all alpha/beta versions? Most of the people forgets to record NMEA

pventafridda, 2015-03-23:
it would hog the system over any limit, and people would forget to disable it..

From: SergioDaSilva82 
Sent: Monday, March 23, 2015 11:35 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LKASSERT Library/PressureSensor.cpp line 38 (#631)

Is it possible to make LK8000 create NMEA automatically in all alpha/beta versions? Most of the people forgets to record NMEA

SergioDaSilva82, 2015-03-24:
That is true but sometimes is so hard to remember to activate the NMEA when we are testing beta or alpha versions.

pventafridda, 2015-03-24:
The error is now managed, and would not lead to crashes out of testbench versions.
However I have open the issue relative to SinkRate which must be checked because the problem is there, and these are only (necessary) workarounds.

pventafridda, 2015-03-24:
Important notice concerning my opening ticket : the crash was NOT due to the error itself, it was due to a diagnostic problem in StartupStore and LKASSERT, in linux only.

closed 2015-03-24 by pventafridda