LK8000/LK8000 #633: SinkRate returning invalid value

pventafridda, 2015-03-24:
Probably under unknown circumstances it is returning an inverted signed value.
We must check .

pventafridda, 2015-04-27:
replicated, problem in Basic Settings (at least)
to replicate:
enter basic settings, not important if flying or not,  change ballast to 100%.
change wing load to 5, change ballast to 0%. crash.

I suspect setBallast is not locking correctly.
Also in dlgBasicSettings all calls to CheckSetBallast are not locked , and the same CheckSetBugs

pventafridda, 2015-04-28:
WeightOffset is becoming negative and a sqrt is performed on it without checking for valid value in GetAUW(). First bug is we are getting back a NaN (not a number) from the sqrt (BETTER TO CHECK ALL sqrt operations in the code).. Why WeightOffset is negative, something to discover tomorrow.

brunotl, 2015-04-28:
weightOffset is calculated when wing load is changed
formula is : wingload * wing area - ballast - massdrygross
so for DG300 with 100% ballast and 5kg/m² load we have : 
5 x 10.27 - 65 - 340 = -353.65

when you change ballast to 0, new gilder weight is calculated ( GetAUW() functions )
formula is : BallastWeight + massdrygross + WeightOffset
so for previously calculated weight offset we have :
0 + 340 + (-353.65) = -13.69

the problem are not in thread locking or anything else, just wrong input value or input in wrong order ...

i don't know were is the solution maybe one of this :
- limit wing load minimum to maxballast / wing area ( 6.33 for DG300 ) 
- don't use ballast for calculate weight offset.
- others ??

pventafridda, 2015-04-28:
good, so we dont really have a problem, only an exception which is badly managed.
I will make the necessary fixes.
The AUW etc is all stuff I did long ago.

From: Bruno de Lacheisserie 
Sent: Tuesday, April 28, 2015 11:59 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] SinkRate returning invalid value (#633)

weightOffset is calculated when wing load is changed
formula is : wingload * wing area - ballast - massdrygross
so for DG300 with 100% ballast and 5kg/m² load we have : 
5 x 10.27 - 65 - 340 = -353.65

when you change ballast to 0, new gilder weight is calculated ( GetAUW() functions )
formula is : BallastWeight + massdrygross + WeightOffset
so for previously calculated weight offset we have :
0 + 340 + (-353.65) = -13.69

the problem are not in thread locking or anything else, just wrong input value or input in wrong order ...

i don't know were is the solution maybe one of this :

  a.. limit wing load minimum to maxballast / wing area ( 6.33 for DG300 ) 
  b.. don't use ballast for calculate weight offset. 
  c.. others ??

pventafridda, 2015-04-28:
The problem is: currently we do allow to set wing loading BELOW the weight reported in the polar.
I am not sure that in such cases the polar is correct.

commit ff40e92c (2015-04-28): Fix #633
closed 2015-04-28 by pventafridda