LK8000/LK8000 #648: Pictorials in waypoint details broken

pventafridda, 2015-04-25:


brunotl, 2015-04-27:
what is the problem ?
[image]

pventafridda, 2015-04-27:
the ptoblem is that terrain painted is fake. not true. invented and 
sometimes not eventi painted
---
Sent from mobile

brunotl, 2015-04-27:
nothing is random, 
background bitmap is just painted with OR operator ( SRCPAINT in GDI) 
so bitmap color is mixed with background color.

pventafridda, 2015-04-27:
Ok I fix it myself.

brunotl, 2015-04-27:
already done

brunotl, 2015-04-27:
why reopen ?

pventafridda, 2015-04-27:
You ask why, I think you did not understand what is the problem.
I prepare some snapshots since you dont check .

pventafridda, 2015-04-27:
ok it is broken in the last v50 too

commit 7277143a (2015-04-27): fix : Waypoint Picto Background Bitmap.
fix #648
closed 2015-04-27 by pventafridda