LK8000/LK8000 #649: waypoint icons not fully transparent on linux

pventafridda, 2015-04-27:
Check on DEMO the bridge "Paderno D ADDA", it is transparent on Windows, but white-filled on linux

brunotl, 2015-04-27:
are you sure ?
[image]

pventafridda, 2015-04-27:
yes. the terrain inside the bridge icon in your picture is white, which is 
wrong. look on 5.0 there is no white
---
Sent from mobile

brunotl, 2015-04-27:
tranparency is defined by mask, so no bug in 5.3
[image]

pventafridda, 2015-04-27:
i made the bridge bitmap and i am saying that it is not painted correctly 
which is 100% sure. i will fix it myself

---
Sent from mobile

brunotl, 2015-04-27:
not related to transparency.
BMP to PNG convertion need to force output format to PNG24 otherwise, grey line are converted to white

brunotl, 2015-04-27:
already done

pventafridda, 2015-04-27:
[image]
this is the correct v5.0.  It is still wrong in master.

brunotl, 2015-04-27:
look your icon, transparency is defined by mask, ( left part of bitmap file
)

if 5.0 use same bitmap, (far i know nobody have replaced) bug are in 5.0 ...

pventafridda, 2015-04-27:
no transparency depends also on what is in the right part.
I remember I had to deal with both . 
5.0 is displaying correctly, and not by random chance.

From: Bruno de Lacheisserie 
Sent: Monday, April 27, 2015 10:34 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] waypoint icons not fully transparent on linux (#649)

look your icon, transparency is defined by mask, ( left part of bitmap file
)

if 5.0 use same bitmap, (far i know nobody have replaced) bug are in 5.0 ...

brunotl, 2015-04-27:
if transparency is defined by white color in right part, why we have mask ? 

so, we have to option : 
 - use only mask for transparency and update bridge bitmap in this way 
 - remove all mask from all bitmap and use white for transparency

second method have advantage of less cpu usage, no need to copy mask & image ...

pventafridda, 2015-04-27:
The bitmap is wrong. wait

From: Bruno de Lacheisserie 
Sent: Monday, April 27, 2015 10:45 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] waypoint icons not fully transparent on linux (#649)

if transparency is defined by white color in right part, why we have mask ? 

so, we have to option : 

  a.. use only mask for transparency and update bridge bitmap in this way 
  b.. remove all mask from all bitmap and use white for transparency 
second method have advantage of less cpu usage, no need to copy mask & image ...

pventafridda, 2015-04-27:
The bitmap you show is not correct.
In v50 we have another.
[image]

brunotl, 2015-04-27:
where is this bitmap ? 

this is change log of this bitmap ( hight and low res ): 

https://github.com/LK8000/LK8000/commits/master/Common/Distribution/LK8000/_System/_Bitmaps/BRIDGE.BMP
https://github.com/LK8000/LK8000/commits/master/Common/Distribution/LK8000/_System/_Bitmaps/BRIDGE_H.BMP

pventafridda, 2015-04-27:
I am wrong that bitmap was an old one.
The current bitmap is the one you reported.

From: Bruno de Lacheisserie 
Sent: Monday, April 27, 2015 11:04 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] waypoint icons not fully transparent on linux (#649)

where is this bitmap ? 

this is change log of this bitmap ( hight and low res ): 

https://github.com/LK8000/LK8000/commits/master/Common/Distribution/LK8000/_System/_Bitmaps/BRIDGE.BMP
https://github.com/LK8000/LK8000/commits/master/Common/Distribution/LK8000/_System/_Bitmaps/BRIDGE_H.BMP

pventafridda, 2015-05-12:
I fixed the bitmaps to paint transparently inside the bridge, as it should. New bitmap in next commit.

commit 15e98df7 (2015-04-27): fix BMP to PNG convert : png need to be 8bit deph for avoid color artefact.

fix : #649
closed 2015-05-12 by pventafridda