LK8000/LK8000 #658: Not saving options?

SergioDaSilva82, 2015-05-05:
Paolo,
I don't if is a bug or the new way LK8000 will hand certain options but I got this report of a pilot using Kobo with all the commits up to May 4th.

only tested in the SIM mode: 

1. In the past after the start glider position on the map was over the "home waipoint" - now after restart glider position is in the middle of the map 
2. I am using only the "variobar" on the main screen - now after each restart I have to swich two times overlays to do not display overlays and display only the variobar 
3. FAI traingle display have to be activated everytime after restart of the LK8000

pventafridda, 2015-05-05:
New way see commits
---
Sent from mobile

closed 2015-05-05 by pventafridda

SergioDaSilva82, 2015-05-05:
I missed that commit! thank you so much ;)

pventafridda, 2015-05-05:
You are much welcome!
I wonder if you could make the “CHANGES” somewhere updated..
Truly we should do it ourselves but we have to catch up with all the changes so far now, and it is quite long work..
I prefer to work on the UI.

From: SergioDaSilva82 
Sent: Wednesday, May 06, 2015 1:49 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Not saving options? (#658)

I missed that commit! thank you so much ;)

SergioDaSilva82, 2015-05-05:
I can try to do that, I usually keep eye on the new features but I really
missed that I don't know how.

I will see if I have sometime on the weekend.

pventafridda, 2015-05-05:
yesss please go backward in commit history, you can use the wiki section on github , it is then easy to take the changes and put them in a text file when we do a release..
And we could all partecipate to prepare the docs in this way!

From: SergioDaSilva82 
Sent: Wednesday, May 06, 2015 1:54 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Not saving options? (#658)

I can try to do that, I usually keep eye on the new features but I really
missed that I don't know how.

I will see if I have sometime on the weekend.

SergioDaSilva82, 2015-05-06:
Something like this? https://github.com/LK8000/LK8000/wiki/LK8000-Changelog

pventafridda, 2015-05-06:
Yes exactly. Maybe with decent font size!!!! It is forbidden on the wikis to use over 40ppt fonts!! haha

From: SergioDaSilva82 
Sent: Wednesday, May 06, 2015 2:03 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Not saving options? (#658)

Something like this? https://github.com/LK8000/LK8000/wiki/LK8000-Changelog

SergioDaSilva82, 2015-05-06:
where can I change the font size?

pventafridda, 2015-05-06:
i was joking

From: SergioDaSilva82 
Sent: Wednesday, May 06, 2015 2:32 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Not saving options? (#658)

where can I change the font size?