LK8000/LK8000 #671: Issues with task using Kobo

SergioDaSilva82, 2015-05-12:
I got this report but I didn't check it yet:

"When I create a task with 3 or more waipoints (using version 5.3A2 from May 10 on Kobo Mini), start simulation mode, start task in simulation, TaskCovered infobox shows properly the task covered distance in km. After reaching the next waypoint the TaskCovered infobox gets cleared a shows the info NO MORE. Using the same task in PNA version of LK8000 (version 5.0A from May 30, 2014), task is properly being calculated and displayed. Both devices set in paragliding mode the same way."

pventafridda, 2015-05-12:
Does it also says “Hit the road Jack”, before NO MORE?

From: SergioDaSilva82 
Sent: Wednesday, May 13, 2015 12:17 AM
To: LK8000/LK8000 
Subject: [LK8000] Issues with task using Kobo (#671)

I got this report but I didn't check it yet:

"When I create a task with 3 or more waipoints (using version 5.3A2 from May 10 on Kobo Mini), start simulation mode, start task in simulation, TaskCovered infobox shows properly the task covered distance in km. After reaching the next waypoint the TaskCovered infobox gets cleared a shows the info NO MORE. Using the same task in PNA version of LK8000 (version 5.0A from May 30, 2014), task is properly being calculated and displayed. Both devices set in paragliding mode the same way."

SergioDaSilva82, 2015-05-12:
I have no idea :) Could you replicate it?

pventafridda, 2015-05-19:
I could not find "NO MORE", or "no more" or "No More" anywhere in the code or languages!

pventafridda, 2015-06-13:
I suspect a language problem file in the user configuration. What language is he using? I cannot confirm this problem as a software problem, since there is no "no more" in the code!

closed 2015-06-13 by pventafridda

SergioDaSilva82, 2015-06-13:
I ask him for more info and I got no reply. Sorry.