LK8000/LK8000 #672: Kobo & Digifly

SergioDaSilva82, 2015-05-13:
Hi, 
Thank you for project LK8000 on Kobo development. I am testing LK8000 with vario transmiting data in Digifly Leonardo protocol. There is a problem. LK8000 does not show HBAR. Pressure indications from vario get to LK8000. It is visible in added NMEA log. GPS and vario works well. IN LK8000 program as a device is choosen Digifly. With the same hardware XCsoar works well and HBAR shows properly. I add Digifly Leonardo protocol's specification, which I found. Please fix this mistake. Thank you ever so much. 

Sincerly, 

Vytautas 

https://www.dropbox.com/s/27r32nny9ooi04d/digifly.zip?dl=0

pventafridda, 2015-05-13:
this is NOT the place for support to end user.
The forum is the place.

closed 2015-05-13 by pventafridda

SergioDaSilva82, 2015-05-13:
He was reporting a bug. I will post in the forum. Sorry

pventafridda, 2015-05-13:
I am not going to check if the user has disabled baro altitude.

From: SergioDaSilva82 
Sent: Thursday, May 14, 2015 12:58 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Kobo & Digifly (#672)

He was reporting a bug. I will post in the forum. Sorry