LK8000/LK8000 #675: linux critical problem with UTF8 waypoint names

pventafridda, 2015-05-23:
LK8000-LINUX_debug: Common/Source/xcs/Screen/Custom/MoreCanvas.cpp:72: const PixelSize Canvas::CalcTextSize(const TCHAR*, size_t) const: Assertion `ValidateUTF8(duplicated)' failed.

obtained either printing the waypoint name on map, or by NAV Waypoint lookup .

Test .cup file:
name,code,country,lat,lon,elev,style,rwdir,rwlen,freq,desc
"Carrè Alto",CAREALTO,IT,4606.467N,01035.750E,3462.0m,7,,

(normal text file under windows, working on Windows no problem.
Attention: on non-debug version, a Segmentation fault is generated with no warning.
The code from xcs is not fault tolerant!

pventafridda, 2015-05-23:
If file is encoded in UTF-8 unicode with BOM (ef bb bf) no problems.
If cup file is ISO 8859, crash

brunotl, 2015-06-02:
duplicated : #676

closed 2015-06-05 by pventafridda