LK8000/LK8000 #676: Support ISO 8859 or go back to UNICODE

pventafridda, 2015-05-24:
Loading text from a file written on a PC (the standard) or importing a waypoint file, a notepad file, anything like a text file, will not necessarily be encoded in UTF8. 
In such case the current implementation from xcs will miserably crash after useless assertions.

brunotl, 2015-05-24:
I Will fix that when i'm back

pventafridda, 2015-05-24:
Great thanks. If you agree, I would at the same time fix the xcs code (like we did for freetype) because it is unacceptable that we have a crash only because of a malformed waypoint file (for example).
I also would much prefer that such cases are not going under the useless "DEBUG" mode, which works only when you can replicate the problem! 
So my intention is to redefine assert into LKASSERT which requires LKAssert.h to be added to the xcs code.

brunotl, 2015-05-27:
Don't do that please !!!

pventafridda, 2015-05-27:
Ok
---
Sent from mobile

closed 2015-06-02 by pventafridda