LK8000/LK8000 #678: Redesign MM bitmaps

pventafridda, 2015-05-25:
40x40  squared, not circled

SergioDaSilva82, 2015-05-25:
Now that you are working on this I would recommend to use letters instead of numbers. Sometimes I forget what is the meaning of each number
M = Alternative map
T = Target
A = Airspace
V = VisualGlide

pventafridda, 2015-05-25:
uh, well..  I dont think  we have to change numbers with letters for multimaps. 
People got used to numbers that represent the sequence .
The most important change will be the MM0 (the main map) and the icons around cruise final glider that are plain misunderstood right now.
We have to test, and adjust. But first I need to make the code that calculates the positions based on icon sizes, then we change the icons.
did you test the page 12 fonts?

SergioDaSilva82, 2015-05-25:
yes,  I did! I like the fact we get about 4 fields on the bottom bar when
using the max.  size. Well done.

I just need to test it a little bit more

SergioDaSilva82, 2015-05-26:
Do you think is possible to add the same options for the bottom bar to the Map overlay text? I think will be better than big and normal

pventafridda, 2015-05-26:
it takes time, currently I had a few months ago fixed the overlays to a decent auto-resolution point.
I must check to see how big work it is.

From: SergioDaSilva82 
Sent: Tuesday, May 26, 2015 2:06 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Redesign MM bitmaps (#678)

Do you think is possible to add the same options for the bottom bar to the Map overlay text? I think will be better than big and normal

closed 2015-05-26 by pventafridda