LK8000/LK8000 #680: Configurable overlays in page 12

pventafridda, 2015-05-27:
The new rescaling engine is half of the work done. New overlays will be rescalable as well, maybe with the exception for time gates for paragliders (lots of informations tightly tuned to fit in little space, no reason to make it better).
We shall have thus dedicated fonts for overlays and abandon the big-small.
If the work proves to be beyond a reasonable time I will suspend it, no warranty.

pventafridda, 2015-05-28:
good news, it comes out quite nicely. we are abandoning big font-small font for overlays, and have an 11 levels choice. Really worth while!

SergioDaSilva82, 2015-05-30:
Well done! I was waiting for this for a while.

pventafridda, 2015-05-30:
I need some more time to fix it.. quite long work.

From: SergioDaSilva82 
Sent: Saturday, May 30, 2015 8:45 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Configurable overlays in page 12 (#680)

Well done! I was waiting for this for a while.

closed 2015-06-02 by pventafridda