LK8000/LK8000 #688: Terminal/Monitor for COM ports

SergioDaSilva82, 2015-06-09:
I think is really useful if you are using LK8000 for Kobo with GoFly V4, for example.

pventafridda, 2015-06-13:
The question is "Why"? I need to understand what is the purpose on Kobo to have a terminal monitor.
Please provide some examples, with text to send out - get back in etc.

SergioDaSilva82, 2015-06-13:
For example, can be good to know what type of sentences the Devices are sending, easy to check in real time instead have to read NMEA log. Good to troubleshoot, for example

and also to setup certain devices like GoFly Vario and BluFlyVario, most popular "homemade" varios around.

pventafridda, 2015-06-13:
So far we have been dealing with Flarm setup (done via serial, like for GoFly) using buttons, in case you want to quickly add some config.

From: SergioDaSilva82 
Sent: Saturday, June 13, 2015 11:24 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Terminal/Monitor for COM ports (#688)

For example, can be good to know what type of sentences the Devices are sending, easy to check in real time instead have to read NMEA log. Good to troubleshoot, for example

and also to setup certain devices like GoFly Vario and BluFlyVario, most popular "homemade" varios around.

SergioDaSilva82, 2015-06-13:
To be honest XCSoar already have it done. I think a copy paste to LK8000 will be enough, no much more work.

pventafridda, 2015-06-13:
I ask Bruno what he thinks about this cut and paste "no much work".

SergioDaSilva82, 2015-06-13:
I tried to say, you don't need to start from scratch :)

SergioDaSilva82, 2015-06-14:
http://git.xcsoar.org/cgit/master/xcsoar.git/tree/src/Dialogs/PortMonitor.cpp?id=bb72f2ca071f9a62ec81a922945ac9022485b34f

#include "PortMonitor.hpp"
#include "Screen/SingleWindow.hpp"
#include "Screen/TerminalWindow.hpp"
#include "Screen/Layout.hpp"
#include "Form/Form.hpp"
#include "Form/ButtonPanel.hpp"
#include "Device/Descriptor.hpp"
#include "Util/Macros.hpp"
#include "Util/FifoBuffer.hpp"
#include "Language/Language.hpp"
#include "Operation.hpp"
#include "Thread/Notify.hpp"

closed 2015-06-30 by pventafridda