LK8000/LK8000 #690: Snail Trail shape

lemonydes, 2015-06-11:
1a) change the snail trail to the dost, bigger dots higher lift smaller dot sink, normal dot standard gliders sink. (problem from this solution - during thermalling the big DOTs will cover a lot of space) 
1b) change the snail trail to the dost, more density of the dost in the defined length higher lift, less dost in the defined length of the trail higher sink (problem from this solution - during thermalling there will be mess with the dots) 

2) different shapes / symbols for the sink / list... dashed line normal sink, small crosses hhighier sink, big squares higher lift. (or something simillar). 

3) different width of the line, higher lift -> bigger lift 

And the last one about which I think is best for use and will lead only to the small change in the programming code> 

4) limit the colour of the rainbow for the snail trail. Do not go from the Blue to the Red but use only the light colour for the sink and dark colour for the lift (eye will always focus on the more contrast colour) -> do not invent what is already invented learn from the current solution, for example on the Flymaster devicies you can have snail trail only with the 4 levels of the gray. Black means lift, light gray means sink.

pventafridda, 2015-06-12:
Your suggestion concerns the dithered greyscale version for kobo, correct?

lemonydes, 2015-06-12:
Yes. It is suggestion list for the KOBO version. It is copy from the discussion and I missed first line of the comment. Thank you in adanvced for the ocnsideration.

SergioDaSilva82, 2015-06-12:
Are you talking about something like this?
[image]

pventafridda, 2015-06-12:
The solution n.4 is the easy way for greyscaled screens.
In the picture above the thermal helper of another software, which I personally have been using without success for two years before developing the LK solution (colored path, we were the first software to adopt it).

SergioDaSilva82, 2015-06-13:
Thank you for working on this ;)

Can FAI triangle be next? https://github.com/LK8000/LK8000/issues/616

pventafridda, 2015-06-13:
I dont think there is a good solution for the lines of FAI triangles. Either they can be seen or not. And to be seen they should be all dark grey or black.

commit e988e951 (2015-06-13): Greyscale trail drawing. Fix #690. The b&w trail is: black color for lift, light grey for sink.
The sink is always a DASHED line, easy to understand. DASH=SINK.
Then, the thickness of lift and sink lines represent the intensity of netto vario as usual.
Important: since we are drawing greyscale, the underneath terrain should be disabled at least in thermal.
A possible next step is to reduce the length of the trail to -say- 5 minutes instead of 10 as now, in order to reduce the sc
reen cluttering.
closed 2015-06-13 by pventafridda