LK8000/LK8000 #693: More overlays control. (On/Off)

lemonydes, 2015-06-15:
I would like to have more possibility to control about Overlays informations. Basicly at least possibility to switch on/off some of them not only half/full.

Will be posible to divede the overlays informations to four groups?

1) Target
2) Wind
3) Left side (Height, Speed, Vario)
4) Right side (Mc, Glide ratio, Height of the arrival)

Reasons: Fling paraglider not for the competition and no need to reach the airport as in the glider i found many of the informations from the overlays just not usufull (and it is not only me, but also my locacl group)

Ad1) Target - during the free flight not usable
Ad2) Wind - wind information is tentersting but we can see it clearly marked with the Arrow and the number on the arrow, this information is much more usefull
Ad3) One of the most usefull information everybody will use it, it is pity that is showed as the last option
Ad4) Most of the paragliders have no glue how to use Mc numbers and also from the theoretical point of view using this values on the paraglider in the mountains is qeustionable, Glide ratio is fine, but also not critical and height of the final arrival in to so interesting during the flight with the no targets.

My intention is to make the map and informations more clear. With the great adjustable bottom bar, some of the informations have no need to be displayd most of the time. :+1:

pventafridda, 2015-06-15:
Sure we can work it out.
In Engineering mode it is already possible to disable MC overlay.
But before proceeding, we must agree on a good solution.
On-Off, is the basic choice, and we can make it all the way.
Lets try to elaborate a bit more, what else can we do to make it better, apart to disable things?
Change items?

pventafridda, 2015-06-15:
Lemon, I dont have a way to get in contact with you with email, to discuss this.

SergioDaSilva82, 2015-06-16:
Just don't disable the Time Gates on map overlays, no matter what they should be there always when a Task is setup. Don't you agree?

I was thinking about a new sub-configuration page for Screen Data on system setup "Map Overlays" with the option to change Fields from Left, and Right side and also option to Wind (on/off).

pventafridda, 2015-06-16:
we cannot change all fields. And to make them configurable is a hell of a work.
Some fields like MC and arrival altitude cannot be moved from their position because of the safety boxes near to them. There is no much space available, everything is tightly disposed at best right now. 

I think that Target overlaymust be always there. It must always show what the rest of values are relative to (arrival, glide ratio required, bearing). If people do not need Target, then they do not need also all glide computer values. 
We may have a simple configuration like vario, hgps and speed. But, when you then read all values in bottombar relative to a target, you will never know what target it is, without having an infobox showing it. Rules out, too complicated and misleading.

We can have all other values On-Off like for bottombar, I will do it quickly soon.
The right values can be changed with something else (not user configurable), but why? The glide ratio is an important information. The arrival altitude to the target too.
The MC is probably the best candidate for a change. Only, the speed to fly and arrival altitude are all relative to this value.

On a different level, I think as you said once, we should instead of MC adopt something completely different, a speed indicator for paragliders based on "slow, medium, fast". Because the problem is that you cannot regulate the speed on a pg, can you?  What are the speed levels we can use in real flight?
Supposely they are only 3, then we can assume the polar of a pg to be on only 3 points, shifted by ballast of course. No MC, but M1, M2, M3. Does this make sense? I understand Bruno is the man here, I am only placing a question.

brunotl, 2015-06-16:
my point of view : 

- target :  is not useless even in freeflight, if you click on, you change Target to LastTermal/TakeOff/nothing or alternate if you have set turmpoint like A/R close point

- wind : it's point of view, personaly i prefert to see  "SSO/15" in big font instead of small arrow ...

- Height/Speed : no discuss about that.

- Vario / Speed to fly :  its already configurable, you can choose Vario/STF/netto

 - Glide Ratio : usefull, no reason to loose time and add another parameters for that

 - Arrival Height : alredy hidden if no Target, so no need to change.

 - MC : usefull if configured to "MC auto thermal" even if you don't really understand MC theory.
and i think all xc pilot need to understant the base of MC theory instead of hide it ...

in all case : most users complain about number of parameters so i think it's important to not add anothers if it's not really needed

pventafridda, 2015-06-16:
i 100% agree. (We can reduce the number of items by placing them in submenus. The on-off is wanted also by GA people.)

lemonydes, 2015-06-16:
ON/OFF will be nice as I am using LK8000 also on my paramotor flying to avoid air space since i have GoFfly V4 from Jarek. Using of the GA scheme is not OK for paramotoring due to the "take off speed" setup for the km/h. Starting to the headwing and continuing the flight to the wind leads to not detection of the start. Therefore the paragliding setup is nedeed.

brunotl, 2015-06-18:
well, so good way is to fix GA scheme for allow to use in paramotor  instead  of add few another parameters in too long list of parameters ?

lemonydes, 2015-06-19:
Was testing yesterday with the GA scheme. And for now paragliders scheme is much closer for the great use than GA.

- In GA take off speed is fixed to 40kmh
- when displaying vario values on the overleys there is also glide ratio value which for 95% time of the flight it is "oo"
- bearing circle indicator is going through  the vario and glide ratio indicators.
- FAI traingles and competition caluclation is not working in the GA (or I do not find the way how to activate it :-/ )

Still fings that hiding some overlays in the PG scheme will be much more easier. (For example on the font size menu, put one more value with the OFF)....

pventafridda, 2015-06-19:
I dont know what you are referring to, Bruno, when you say "adding another parameter". 
Lemon: ok we got the point, takeoff is 40kmh and your aircraft is neither a paraglider nor a motorplane, so use what you like. The overlays will come to be configurable in v6.

SergioDaSilva82, 2015-06-22:
@lemonydes Can you try the new version I uploaded? https://www.dropbox.com/sh/tushx43wde0gn5v/AAAkOv5IKf97JoCkuojrgKP1a?dl=0

lemonydes, 2015-06-23:
Thank you :+1:  :+1:  :+1:  solution exceeds my expectation, great work.

PS: I suppose there is only minor bug in the setup. "Right Bottom" value you can choose AUX3 for the output. But the value from the AUX2 field is shown.

closed 2015-06-23 by pventafridda