LK8000/LK8000 #718: Crash after takeoff

pventafridda, 2015-07-05:
Platform: PNA
Replicated: some times, yet not sure how to make it at will

SIM mode, PAN mode, drag to make the aircraft move (see other bug). After TAKEOFF sometimes we get a crash with no diagnostics.

To be investigated in all platforms, could not be related to SIM, and-or to PAN.

SergioDaSilva82, 2015-07-05:
It happened to me twice, I replay demo IGC file and I went to analysis FAI page and when I heard take off sound and after that the buttons were not working, I waited 2 minutes and then was working again.
This was tested with Kobo. I never reported because I couldn't replicated it again.

pventafridda, 2015-07-05:
Anything is useful to track the problem. Please always report a crash, even if you cant replicate it.
Now the best move is to make it replicate at will , and give the procedure to do it. 
So we know this is not related to PNA, it is a general problem. Apparently related only to SIM mode, but we cannot be sure (I would bet on it though).

SergioDaSilva82, 2015-07-05:
And I think one of the times was crashed when detected free flight.

SergioDaSilva82, 2015-07-05:
For a few moments I thought was related to this fix: https://github.com/LK8000/LK8000/issues/635
but I don't think so.

pventafridda, 2015-07-05:
I dont think it is related . It should be an issue inside the Calc thread. 
If – as I suspect – it happens only in SIM mode while moving (or replaying an IGC, same thing), then it is a missing lock .
But the interesting thing is to know why we get the crash without an LKASSERT error. 
We have 590 sentinels inside the code, and I wish we had twice as much.
If we have a crash on PNA, without LKASSERT we have zero chance to fix it without spending weeks or months.
Sometimes we have the crash on PNA but not on PC or Linux, simply because of different architecture and memory management.
We still have the bug on all platforms, but it appears only on PNA, as an example. CE is more vulnerable due to tight memory.
Last year, v5 was delayed by three months for this reason.
Anyway, first thing is to replicate. Once replicated, we can easily fix it. It is a good news to know that it happens on linux also.

From: SergioDaSilva82 
Sent: Monday, July 06, 2015 1:24 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Crash after takeoff (#718)

For a few moments I thought was related to this fix: #635
but I don't think so.

pventafridda, 2015-07-07:
Sergio do you confirm on Kobo it CRASHED, or did it simply hang (not taking inputs) for a while?
A crash will make it exit, of course. No recovery.

SergioDaSilva82, 2015-07-08:
Sorry. It was a hangup and not a crash.

pventafridda, 2015-08-15:
It was the airspace warning problem!

closed 2015-08-15 by pventafridda