LK8000/LK8000 #725: grahics glitch in Linux-Version

klinkenberg, 2015-07-17:
There is graphics glitch within the Linux-Version 5.3e0 which was introduced into the Linux-versions sometimes in spring.
The appended pictures show the map viewing differences in the very flat lands, especially the "flatland mountains"  (height differences in reality some tens meters).

[image]
[image]

pventafridda, 2015-07-17:
thanks ernst, problem is well exposed.

brunotl, 2015-07-28:
what is terrain & topology file ?

klinkenberg, 2015-07-28:
> what is terrain & topology file ?

Hallo Bruno,

give me please a more capable email-adress I can send the files to. Github is 
restricted to somewhat 4MB 

Greetings Ernst-Dieter

> 
> 
> ---
> https://github.com/LK8000/LK8000/issues/725#issuecomment-125513281

pventafridda, 2015-07-29:
awaiting for ernst to confirm it is fixed

closed 2015-07-30 by pventafridda

klinkenberg, 2015-08-28:
I confirm, Bug is fixed

commit b4094f64 (2015-07-30): TerrainRenderer::FillHeightBuffer, fix positive altitude check, unsigned short are always positive, so compare need to be done before unsigned cast

fix #725