LK8000/LK8000 #729: multiselect not working in multimaps

pventafridda, 2015-07-30:
reported by ulli

brunotl, 2015-07-30:
tested on all multimap, all work on PNA ...

pventafridda, 2015-07-30:
Oh interesting.  I tested on V1 which is CE 6. Ulli has the same. It does 
not work there.
I will check it works on CE5.

Maybe it is the same problem of poco gettick.
---
Sent from mobile

brunotl, 2015-07-30:
tested on WM6.5

pventafridda, 2015-07-30:
That is CE 5
---
Sent from mobile

pventafridda, 2015-07-30:
I tested on V2, it works (CE 5). I tested on V1, CE 6 does not work. It must be the poco library. no problem

pventafridda, 2015-07-31:
It is the LKEVENT_LONGCLICK  not correct in CE 6.

brunotl, 2015-08-04:
work fine on  CE 6.00 (build 2217)

pventafridda, 2015-08-04:
On emulator ?
---
Sent from mobile

brunotl, 2015-08-04:
No Bug.
with correct "_System\DEFAULT_MENU.TXT" file, all work fine on V1.

pventafridda, 2015-08-05:
I confirm.

closed 2015-08-05 by pventafridda