LK8000/LK8000 #734: Dialog not working correctly

pventafridda, 2015-08-09:
dlgAirspaceDetails xml has not changed since v5, but the rendering is now different.

Previously:
[image]

Now:
[image]

Furthermore, we are now painting the background map that is not updated (clock is stopped).

pventafridda, 2015-08-09:
The color of unselectable buttons (like Top Base Range etc.etc.) should be darker to mean "disabled".

closed 2015-08-10 by pventafridda

AlphaLima, 2015-08-11:
Works for Linux now, but still not for windows.
Tested with an complete updated installation.

pventafridda, 2015-08-12:
I confirm it was fixed, the rendering was correct.
The button not working were another issue.

commit 6773b344 (2015-08-10): fix AirspaceDetail dialog layout

Fix #734