LK8000/LK8000 #740: Strange farvisibles waypoints number

pventafridda, 2015-08-13:
Normally farvisibles are around 50-60..
Reminder to check.

3098489113] ... ** UpdateActiveScreenZone 0,0,536,724
[3098489117] ... Unload Fixed Bitmaps
[3098489118] ... Unload Profile Bitmaps
[3098489118] ... Loading Fixed Bitmaps
[3098489125] ... Loading Profile Bitmaps
[3098489177] .... CreateCalculationThread
[3098489228] .... ProgramStarted=InitDone
[3098489279] ... DrawThread START
[3098489330] ... BOTTOMBAR splitter=4
[3098489332] ... LKBatteryManager waiting for first update
[3098489745] . CloseProgressDialog
[3098489745] . GCE_STARTUP_SIMULATOR
[3098489745] . Create default task
[3098489745] . DefaultTask assigning Home (wp=2390) as default destination
[3098489752] . ProgramStarted=NormalOp h2:05:53 (UTC 00:05)
[3098489752] . Free ram=132276 K  storage=4294967295 K
[3098490258] . LoadTask : </root/LK8000/_Tasks/Default.lkt>
[3098490327] ... LKBatteryManager waiting for first update
[3098491298] ... LK BatteryManager V1: internal battery information not available, function disabled
[3098493288] . SIMULATOR: real init
[3098493291] ...... Checked 558 farvisibles waypoints for maxRange=15000.000000, type=0
[3098493292] ...... Checked 558 farvisibles waypoints for maxRange=15000.000000, type=0
[3098493293] ...... Checked 558 farvisibles waypoints for maxRange=15000.000000, type=0
[3098493295] ...... Checked 558 farvisibles waypoints for maxRange=15000.000000, type=0
[3098493297] ... Retuning dstrangeturnpoint to 88 km

pventafridda, 2015-08-15:
it is a timing problem on startup, farvisibles not yet being calculated. Not critical but should be fixed.

brunotl, 2015-08-21:
egression commit : 281935f36607bbcbae8c6696305380aecc829ccf

pventafridda, 2015-11-26:
ehm. does it mean it is fixed with the regression?

brunotl, 2015-11-26:
no, problem come from commit 281935f36607bbcbae8c6696305380aecc829ccf

without "welcome screen", check farvisibles waypoints is done too early...

closed 2016-01-22 by pventafridda