LK8000/LK8000 #744: Increase Airspace details info

SergioDaSilva82, 2015-08-15:
I got this request on PGForum: http://www.paraglidingforum.com/viewtopic.php?t=73703

I'm generating an airspace file, contaiing NOTAMs, so that various devices can display NOTAMS on their screens. So far, it's all fine. I have created a file for import into XCSoar and LK8000. 

XCSoar is brilliant - I can see the NOTAM type, and clicking on it will show the entire NOTAM text. 
However, on LK8000, if I select an area, I get a screen like the attached photo. Clicking on it doesn't show anything worth looking at. 

The Airspace and NOTAMs are listed in a larger font than I'd like . You can see it's a NOTAM, but not much else. 

I'm a total beginner on LK8000 - is there a way of changing that font? 

[image]

[image]

[image]

pventafridda, 2015-08-16:
LK does not consider currently the airspace name as a note information, and thus it does not have space for it, in the airspace details dialog.
We have two more buttons not available in xcsoar.
1. Select . Using this button the airspace item will be elected as the one for which special distances (vertical and horizontal ) are available as infoboxes.
2. Fly In .  This button will revert the paradigm of airspaces: for example, you may be obliged not to exit a certain area. This is a FLY IN area, and you will get warnings concerning outbound directions, instead of the opposite.

For the future we may split the CLOSE button into two buttons, one having “DETAILS” and the other still being CLOSE.
However, the neat solution should be to have a NOTAM class, and display them with a standalone dialog.

I would not suggest to operate changes at this stage before we fix all the problems on PNA.

From: SergioDaSilva82 
Sent: Saturday, August 15, 2015 10:44 PM
To: LK8000/LK8000 
Subject: [LK8000] Increase Airspace details info (#744)

I got this request on PGForum: http://www.paraglidingforum.com/viewtopic.php?t=73703

I'm generating an airspace file, contaiing NOTAMs, so that various devices can display NOTAMS on their screens. So far, it's all fine. I have created a file for import into XCSoar and LK8000. 

XCSoar is brilliant - I can see the NOTAM type, and clicking on it will show the entire NOTAM text. 
However, on LK8000, if I select an area, I get a screen like the attached photo. Clicking on it doesn't show anything worth looking at. 

The Airspace and NOTAMs are listed in a larger font than I'd like . You can see it's a NOTAM, but not much else. 

I'm a total beginner on LK8000 - is there a way of changing that font?

parapenT1sta, 2017-02-11:
I am closing it. Forget about it.

closed 2017-02-11 by parapenT1sta