LK8000/LK8000 #756: Option to enable or disable UnGhost mode

dtykhon, 2015-11-08:
It would be nice to have an option to enable or disable UnGhost mode.
It was discussed on PGF http://www.paraglidingforum.com/login.php?redirect=viewtopic.php&p=p476883#p476883
Thanks

pventafridda, 2015-11-08:
I will make it configurable at different levels 0 to 3 or something like.
---
Sent from mobile

SergioDaSilva82, 2015-11-08:
Thanks Paolo. This is the link about the discussion: http://www.paraglidingforum.com/viewtopic.php?p=p476883#p476883

pventafridda, 2016-02-08:
no reason to change back what had been changed and changed again.

dtykhon, 2016-02-08:
Hello Paolo,
Have you seen the video?

dtykhon, 2016-02-08:
LK8000 on Glo and Glo HD: https://youtu.be/eDTXQFwhyR8?t=50
Both devices blink moderatly.

XCSoar on Glo and Glo HD: https://youtu.be4XR8Ygpt9-g?t=514
Glo HD blinks terrible
 Glo does not blink at all - wow, looks awesome! Is it possible to implement such behavior in the LK8000?

pventafridda, 2016-02-08:
The second video has no correct link.

I am sorry but my idea of unghosting is different from the one implemented in LK.
The main issue with unghosting is : when to do it. 
I wanted an euristical approach, gained through accounting “dirty” frames, and a delay before unghosting while people is turning the pages.

From: Dmytro 
Sent: Monday, February 08, 2016 4:01 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Option to enable or disable UnGhost mode (#756)

Hello Paolo,
Have you seen the video?

dtykhon, 2016-02-08:
Sorry, the correct link for the second video is https://youtu.be/4XR8Ygpt9-g?t=515

brunotl, 2016-02-08:
XCSoar don't have "unghost" feature,  it's only a "blinking" bug on GloHD when screen refresh are too fast.

fixed in LK8000 by commit 9854e2f187c55f19f2ef860187fdcd1ddb62bedb

dtykhon, 2016-02-08:
I think XCSoar (without unghost) looks definitly better tnan LK8000 (with
unghost).
Hope they will fix the blinking bug.
Thank you Paolo!
With warmest wishes,
Dmytro

pventafridda, 2016-02-08:
So I dont understand, maybe you can explain.
What are we missing? By Unghosting we mean the “flashing” effect to clean up the ghosting problem of eink .
Do you mean that GloHD does not have any ghosting problem and there is no need to unghost ..?

From: Dmytro 
Sent: Monday, February 08, 2016 11:40 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Option to enable or disable UnGhost mode (#756)

I think XCSoar (without unghost) looks definitly better tnan LK8000 (with
unghost).
Hope they will fix the blinking bug.
Thank you Paolo!
With warmest wishes,
Dmytro

pventafridda, 2016-02-08:
in the video almost nothing is painted, no terrain, no graphics.. so there is no ghosting effect. It looks strange to me that xcsoar does not do unghosting at all. Certainly for a reason, but in our case without unghosting we have a noisy screen full of background artefacts after a few page flips.