LK8000/LK8000 #790: LINUX only: LANGUAGE problem

pventafridda, 2016-01-26:
See http://www.postfrontal.com/forum/topic.asp?whichpage=2&TOPIC_ID=8103&#67649

Happens on Linux, good example is look at Battery (message 1132).
No need to check on Kobo, same thing. If we fix on linux it is fixed on Kobo.

pventafridda, 2016-01-26:
Msg 1132 is RUS:MSG is encoded (binary) as 
0xd0 0x91 0xd0 0x90 0xd0 0xa2 0xd0 0x90 0xd0 0xa0 0xd0 0x95 0xd0 0xaf 
corresponding to 
_@M1132_ "БАТАРЕЯ"

brunotl, 2016-01-26:
2 problem :

- Battery (message 1132), i don't know where is problem, maybe freetype or dejavu font  versions related ( i can't replicate with DejaVu 2.35-1  and freetype 2.5.2-4 )

- all others are linked to string truncated to Byte count instead of Character count (fix in progress).

pventafridda, 2016-01-26:
probably string truncation, also for battery. I see no other possible reason, since the 0xd0 0x90 character is written correctly in second position.
As a side effect I am learning russian!

From: Bruno de Lacheisserie 
Sent: Tuesday, January 26, 2016 5:20 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LINUX only: LANGUAGE problem (#790)

2 problem :

  a.. Battery (message 1132), i don't know where is problem, maybe freetype or dejavu font versions related ( i can't replicate with DejaVu 2.35-1 and freetype 2.5.2-4 )

  b.. all others are linked to string truncated to Byte count instead of Character count (fix in progress).

closed 2016-01-28 by pventafridda