LK8000/LK8000 #795: Add wifi status and connection to status page

ienders13, 2016-02-02:


ti24horas, 2018-04-18:
Is there a reason why we can't enable wifi and scan for wifi networks on lk8000? It could be on startup page, where you select whether to run kobo, fly, sim and so on. Is so anoying to reboot, go to kobo, enable wifi, hard reboot and enter wifi again.

brunotl, 2018-04-18:
you can Enable/Disable Wifi, you have Wifi On/off button in top right corner of "Config" Menu

implement network scan is too much time consuming.

ti24horas, 2018-08-11:
XCSoar provides this feature. Where can I find tutorials on how to contribute with lk8000? I could create an wifi-config page on main screen for example. Then one could put SSID, auth mode, user and password then we could avoid scanning.