LK8000/LK8000 #799: LT Server port crash

pventafridda, 2016-02-04:
On PC (at least..) clicking on Port configuration of LT server in Pilot setup is crashing LK

SergioDaSilva82, 2016-02-04:
Paolo,

Are you sure about it? I compiled a version with commits until Jan 30, and if I run a new installation, it doesn't crash in FLY and SIM mode.

http://tinyurl.com/lk8000v6alpha

pventafridda, 2016-02-04:
yes I am sure, it happened last night. it doesnt happen now, so it is not an easy problem

From: Sérgio Da Silva 
Sent: Thursday, February 04, 2016 11:17 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LT Server port crash (#799)

Paolo,

Are you sure about it? I compiled a version with commits until Jan 30, and if I run a new installation, it doesn't crash in FLY and SIM mode.

http://tinyurl.com/lk8000v6alpha

pventafridda, 2016-02-04:
it just happened again by clicking on help , in the LT screen

From: Sérgio Da Silva 
Sent: Thursday, February 04, 2016 11:17 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LT Server port crash (#799)

Paolo,

Are you sure about it? I compiled a version with commits until Jan 30, and if I run a new installation, it doesn't crash in FLY and SIM mode.

http://tinyurl.com/lk8000v6alpha

SergioDaSilva82, 2016-02-04:
I can confirm, hitting the Port and then Help it crashes.

brunotl, 2016-02-04:
easy with debugger : found in 5 second ....

another global variable uninitialized...

pventafridda, 2016-02-04:
Looks like HELP button is in the XML but not managed by the code.

From: Bruno de Lacheisserie 
Sent: Friday, February 05, 2016 12:42 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LT Server port crash (#799)

easy with debugger : found in 5 second ....

another global variable uninitialized...

closed 2016-02-05 by pventafridda