LK8000/LK8000 #807: Duplicates in dialog lists (?)

pventafridda, 2016-02-09:
Reported by Ernst: airspace list is showing a duplicated (verified).
And we have twice Disabled in the custom key lists. Maybe unrelated.
Investigating..

brunotl, 2016-02-09:
for Custom Key lists, still we use array index instead of id, we can't remove second.

second "disable" replace removed "Active Map On/Off"

pventafridda, 2016-02-09:
ah good. you are right!
If we change array index, then we must change configuration name, and people must reconfigure .. no reason.

So problems are unrelated.

From: Bruno de Lacheisserie 
Sent: Tuesday, February 09, 2016 5:45 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Duplicates in dialog lists (?) (#807)

for Custom Key lists, still we use array index instead of id, we can't remove second.

second "disable" replace removed "Active Map On/Off"

pventafridda, 2016-02-09:
I think it is a problem in the airspace sorting while looking for underneath airspace.
Matching is not correct, maybe. Anyway it is an airspace issue.

pventafridda, 2016-02-10:
So far I cannot confirm anymore the problem. The only duplicate I got was real .
Same for Ulli. I am declassing the issue.

AlphaLima, 2016-02-10:
Ernst also confirmed using two airspace files with partly identical 
airspace definitions.

closed 2016-02-16 by pventafridda