LK8000/LK8000 #808: Airspace, negative distance shown in multiselection list

pventafridda, 2016-02-09:
Strange. Asking for a multiselection list on the map, I also get some airspaces.
Sometime I get a negative distance value for one of these. 
Selecting the item, the distance is correctly positive.

I noticed this item appears twice in the list, sometimes.

AlphaLima, 2016-02-09:
As far as I remember neg. values mean you are "inside" the airspace.
But inside is separate for vertical and horizontal. So it can mean 
under/above left/right of it.
Only if vert. *and* horiz. are negative you're "inside" the particular 
airspace.

For the multiselect we only show the horizontal distance so neg values 
mean you're under, above or inside the airspace.
IT is a precious info but maybe not documented correctly.

AlphaLima, 2016-02-09:
Maybe it is a better solution to show the abs value and have an indicator
(in under above) but I'm afraid it is a language problem and need much 
more space filling the line.

pventafridda, 2016-02-09:
OK. Can you please check then the duplicates issue? I could verify it happens.
Only for airspaces .. Ernst has a german airspace eventually where it happens

From: Ulli Heynen 
Sent: Tuesday, February 09, 2016 7:30 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Airspace, negative distance shown in multiselection list (#808)

As far as I remember neg. values mean you are "inside" the airspace.
But inside is separate for vertical and horizontal. So it can mean 
under/above left/right of it.
Only if vert. *and* horiz. are negative you're "inside" the particular 
airspace.

For the multiselect we only show the horizontal distance so neg values 
mean you're under, above or inside the airspace.
IT is a precious info but maybe not documented correctly.

AlphaLima, 2016-02-09:
Already did it, cold not find a problem.
Probably the airspace is defined twice.

AlphaLima, 2016-02-09:
Can you send me your airspace file and the particular AS where it happened?

AlphaLima, 2016-02-09:
I had quite often multiple defined airspaces.
Not easy to see but simple to prove. Disable the Airspace if it remains 
it is multiple defined.
If you select the same airspace file for Airspace1 and Airspace2 you can 
emulate the effect for every airspace.

pventafridda, 2016-02-09:
I confirm you are right concerning my example!
Checking Ernst now

From: Ulli Heynen 
Sent: Tuesday, February 09, 2016 8:38 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Airspace, negative distance shown in multiselection list (#808)

I had quite often multiple defined airspaces.
Not easy to see but simple to prove. Disable the Airspace if it remains 
it is multiple defined.
If you select the same airspace file for Airspace1 and Airspace2 you can 
emulate the effect for every airspace.

pventafridda, 2016-02-09:
unfortunately ernst did not open an issue on github for this, so i am sending to you via email details

From: Ulli Heynen 
Sent: Tuesday, February 09, 2016 8:38 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Airspace, negative distance shown in multiselection list (#808)

I had quite often multiple defined airspaces.
Not easy to see but simple to prove. Disable the Airspace if it remains 
it is multiple defined.
If you select the same airspace file for Airspace1 and Airspace2 you can 
emulate the effect for every airspace.

pventafridda, 2016-02-10:
OK it looks intentional, so I am declassing to change request. Not that I want or need to change it, personally. negative stands for out of airspace distance. I would close it.. you tell me.

AlphaLima, 2016-02-10:
It is like this since the very first Multiselect. No complains so far.
I suggest to leave it as it is and add a sentence in the docs.

closed 2016-02-16 by pventafridda