LK8000/LK8000 #812: LDROTARY RESET distance errors

pventafridda, 2016-02-19:
We have a testbench reported from real flight of such problem (self-recovered), which we never had before. Better to investigate a bit.

pventafridda, 2016-02-19:
I think it is related to bad position fix . User reported GPS fix problems yesterday at the same time with both LK and an Oudie, so not hardware-software related.
Leaving this as a reminder for some time. No need to investigate further.

brunotl, 2016-02-19:
only one condition for have this log message
 - moving less than 3 meter or more than 150m in 9 consecutive calculation run ( 9s with valid fix, 45s without valid fix )

pventafridda, 2016-02-19:
I think a gps error in position reporting, which is quite common since ever. But it must be less than 3 m move every second for 9 seconds.
This is quite likely to happen if the position does not change . In the runtime , I see user was thermalling at very low altitude for a long time over the airport.
So in my opinion this is the case of gps problem while at high banking.
It cannot be 150m distance every each second, it would seem like a giant white rabbit haha

From: Bruno de Lacheisserie 
Sent: Friday, February 19, 2016 4:54 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LDROTARY RESET distance errors (#812)

only one condition for have this log message

  a.. moving less than 3 meter or more than 150m in 9 consecutive calculation run ( 9s with valid fix, 45s without valid fix )

brunotl, 2016-02-19:
happen also if you loose fix for 45s ...

pventafridda, 2016-02-19:
Dont we have a testbench report in runtime in such case? we should

From: Bruno de Lacheisserie 
Sent: Friday, February 19, 2016 5:06 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LDROTARY RESET distance errors (#812)

happen also if you loose fix for 45s ...

pventafridda, 2016-03-09:
I keep this up for further tests, just to be sure. I keep having unwanted rotary resets.

pventafridda, 2016-03-22:
It can also happen when crossing the 00:00 UTC due to time difference.
Not a problem, closing the issue.

closed 2016-03-22 by pventafridda