LK8000/LK8000 #815: Screen touch non responsive for a short while

pventafridda, 2016-02-19:
PNA and Kobo, I think it is a general problem. Not yet sure 100%, but at least it happened to me also (and reported by users) even in SIM mode. 
For my case, I was in SIM, PNA. After 3-4 seconds the screen was back responsive. During that time it was updating informations as usual, only key touch was ignored, and not buffered of course.

polachp, 2016-03-07:
Paolo, this happens to me too! OnKobo mini, there are sometimes 30sec and more when the screen in unresposive to touch control.

pventafridda, 2016-03-07:
It would be great if you could find the runtime.log pointing to the session where it happened!

pventafridda, 2016-03-08:
I tried for three hours to replicate it, in the same conditions of last time it happened, but no luck.  Whatever it is, it seem not to be critical. But further testing is required.

pventafridda, 2016-03-22:
It could be related to a dialog such as airspace warning?

de07y, 2016-03-28:
(Sorry for my bad english, I am French)
Hello. I did my first flight last Saturday with LK8000 (2h40): I locked my screen before taking off, then it seems the unlock has been removed successfully as when I tried to push the "M" menu nothing appeared. But when I tried later to push at the middle bottom bar to go to the map screen (I was using  a page without map, full of infoboxes), nothing changed. At the end of the flight, I tried to exit LK8000 but nothing was detected from the touch control. So I had to switch off the Kobo with the physical button.  I have been abled then to retrieve the temporary IGC file: the track has been successfully recorded. But as I was not using the map screen, I suppose there is nothing to do with the airspace warning?.  Cheers. Denis.

de07y, 2016-03-28:
There is the RUNTIME.LOG zip file of my Kobo Glo (Non HD): the problem occurred on the 26 of March in the south of France (fr
[RUNTIME.zip](https://github.com/LK8000/LK8000/files/191720/RUNTIME.zip)
om 15h50 to 18h30 when I switch of manually the Kobo as the touch countrol wasn't detected at all).

pventafridda, 2016-03-28:
Why do you say you locked and unlocked? How did you lock and unlock?
---
Sent from mobile

de07y, 2016-03-28:
Sorry I meant: I used the item "Lock before taking off", but I dont think there is anything due to that item: I am trying the LK8000 there on the Kobo setup ant there is some freeze when I touch the screen. (some touch are not recognized, then after few seconds it is recognized). But on saturday during all the flight, my touch haven't been recognized at all till I manually stop the Kobo.
Denis

pventafridda, 2016-03-28:
The lock screen works until you double click on bottom right corner, as it is said when you lock.
You can try, please, and confirm.. It doesnt unlock after takeoff, ever .

From: de07y 
Sent: Monday, March 28, 2016 6:57 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Screen touch non responsive for a short while (#815)

Sorry I meant: I used the item "Lock before taking off", but I dont think there is anything due to that item: I am trying the LK8000 there on the Kobo setup ant there is some freeze when I touch the screen. (some touch are not recognized, then after few seconds it is recognized). But on saturday during all the flight, my touch haven't been recognized at all till I manually stop the Kobo.
Denis

de07y, 2016-03-28:
Thanks for your message: yes it is ok for me:  :-)

Ok, yes it is written in french "Ecran verrouillé jusqu'au décollage"
"Double Clic en bas à droite pour Deverrouiller"

(Screen locked till takeoff , double click on the bottom right side to unlock")
Then I did it and it is written "unlocked".
And This is ok.
But it wasn't this weekend:
I just removed the screws of the Kobo few minutes ago (GoFly version) I probably put the screws too strong last time and now it is ok, it looks Hardware problem because of me: so for me it looks ok (but for others it is perhaps another problem).

Cheers,  :-)
Denis

pventafridda, 2016-03-28:
Please use the support forum for anything not related to development. Do not use github for support. 
Thanks!

From: de07y 
Sent: Monday, March 28, 2016 11:19 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Screen touch non responsive for a short while (#815)

Thanks for your message: yes it is ok for me: :-)

Ok, yes it is written in french "Ecran verrouillé jusqu'au décollage"
"Double Clic en bas à droite pour Deverrouiller"

(Screen locked till takeoff , double click on the bottom right side to unlock")
Then I did it and it is written "unlocked".
And This is ok.
But it wasn't this weekend:
I just removed the screws of the Kobo few minutes ago (GoFly version) I probably put the screws too strong last time and now it is ok, it looks Hardware problem because of me: so for me it looks ok (but for others it is perhaps another problem).

Cheers, :-)
Denis

de07y, 2016-03-29:
I thought it was related to this bug, and then I thought it was answering your question ("It could be related to a dialog such as airspace warning?") as I got this problem on another page than map . Then I gave you the REPORT.log file as I thought it was important. And THEN I saw myself it was due to a hardware problem: then I told you it to avoid any confusion with this problem.
Regards,
Denis

pventafridda, 2016-03-31:
I can finally replicate it on linux, not all times yet. But it is a step forward.

pventafridda, 2016-03-31:
First thing to check is if the global FontRenderer  is not overflowed over LockModeStatus .
not tonight..

pventafridda, 2016-04-08:
Apart from fixing fake repetitions, I could not advance on this matter further because I cannot replicate the problem. If there is a problem really. It is no more a software bug for me until counter proof.

closed 2016-04-08 by pventafridda