LK8000/LK8000 #817: LKASSERT in Geoid.cpp

pventafridda, 2016-02-19:
Strange.

unsigned char* egm96data = NULL;
extern HINSTANCE _hInstance;
#else
#include "resource_data.h"
const unsigned char* egm96data = NULL;
#endif

bool OpenGeoid(void) {
    if(UseGeoidSeparation) {
#ifdef WIN32_RESOURCE
        LKASSERT(!egm96data);
        egm96data = NULL;

pventafridda, 2016-02-20:
[000029081] . [000029079] Starting LK8000 v5.3h Feb 11 2016 PNA
[000029239] . Built with mingw32 3.16 (GCC 4.6.3) 
[000029255] . TESTBENCH option enabled
[000029270] . USELKASSERT option enabled
[000029673] . Globals_Init
[000029687] ... FindFreeSpace <\Storage Card\LK8000\> start
[000029698] ... FindFreeSpace ok
[000029707] . Free ram=59156 K  storage=1830528 K
[000029716] . Host and Cpu informations not available
[000029795] ... Searching modeltype inside default profile <\Storage Card\LK8000\_Configuration\DEFAULT_PROFILE.prf>
[000029810] ... Modeltype not found in profile, probably Generic PNA is used.
[000029820] . Program execution path is <\Storage Card\LK8000\>
[000029829] . Program data directory is <\Storage Card\LK8000\>
[000029840] . Install/copy system objects in device memory
[000029851] . Welcome to InstallSystem v1.2
[000029860] . InstallSystem source directory <\Storage Card\LK8000\_System> is available
[000029880] ------ InstallSystem PROBLEM: cannot locate the Windows folder, got string:<>
[000029890] ------ InstallSystem attempting to use default "\Windows" but no warranty!
[000029899] . Special RegKey for fonts not found on this PNA, using standard folder.
[000029909] ------ PROBLEM: cannot locate the Fonts folder, got string:<>
[000029919] ------ Attempting to use directory <\Windows> as a fallback
[000029930] . InstallSystem: Copy/Check Fonts from <\Storage Card\LK8000\_System> to <\Windows>
[000029945] . Checking TAHOMA font
[000029962] . Font TAHOMA.TTF is already installed
[000029972] . Checking TAHOMABD font
[000029984] . Font TAHOMABD.TTF is already installed
[000029994] . InstallSystem completed OK
[000030005] . InstallSystem ended, code=0
[000030260] . Language load file: <\Storage Card\LK8000\_Language\ENG_MSG.TXT>
[000031061] ... ProfileResetDefault
[000031200] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_AIRCRAFT.acf>
[000031228] ... LKProfileInitRuntime
[000031247] . No special regsets for this device
[000031260] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_PILOT.plt>
[000031277] ... LKProfileInitRuntime
[000031288] . No special regsets for this device
[000031298] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_DEVICE.dvc>
[000031314] ... LKProfileInitRuntime
[000031325] . USING KEYBOARD TYPE 3
[000031336] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_PROFILE.prf>
[000031409] ... LKProfileInitRuntime
[000031438] . USING KEYBOARD TYPE 3
[000031453] . StatusFileInit
[000031466] Init Bluetooth
[000031490] Unable to Load <BTDrv.dll>
[000031517] Bluetooth Is OK
[000031537] Bth (Software\Microsoft\Bluetooth\Device) : Bluetooth Keyboard (BT:989140318861)
[000031549] Bth (Comm\Security\bluetooth) : 001bdc0f5fdf (BT:001bdc0f5fdf)
[000031558] Bluetooth : Device Count = 2
[000031567] ..... WindowResize RECT (rltb) 272 0 0 480
[000031578] . Create main window
[000031623] ..... ScreenGeometry type: 4
[000031638] . InitLKScreen: 272x480
[000031651] . Init_Fonts_1
[000031671] . Init_Fonts_2
[000031723] ... Loading Fixed Bitmaps
[000032314] ... Loading Profile Bitmaps
[000032376] ... LKObjects_Create
[000032410] . Init_Fonts_1
[000032430] . Init_Fonts_2
[000033062] . Init Calculations
[000033073] . Reset flight statistics
[000033125] . Startup dialog, RUN_MODE=0 
[000035887] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_PILOT.plt>
[000036208] ... LKProfileInitRuntime
[000036219] . USING KEYBOARD TYPE 3
[000036227] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_DEVICE.dvc>
[000036242] ... LKProfileInitRuntime
[000036252] . USING KEYBOARD TYPE 3
[000036261] ... LoadProfile <\Storage Card\LK8000\_Configuration\DEFAULT_AIRCRAFT.acf>
[... 77 lines cut ...]
[000060213] .... RMC date field empty, skip sentence!
[000061259] .... RangeTurnepoint=6/500 (TPrange=100) RangeLandable=205/500 (LNrange=150) retunecount=0 retrycount=0 sinceretry=0s 
[000062213] .... RMC date field empty, skip sentence!
[000063435] .... RMC date field empty, skip sentence!
[000064427] .... RMC date field empty, skip sentence!
[000065203] .... RMC date field empty, skip sentence!
[000066203] .... RMC date field empty, skip sentence!
[000067203] .... RMC date field empty, skip sentence!
[000067855] . V7
[000067867] . V7 Serial Number 347
[000067876] . V7 Software Vers.: 3.10
[000067885] . V7 Hardware Vers.: 13.00
[000068203] .... RMC date field empty, skip sentence!
[000068571] ... GPS baro source back available h11:31:22 (UTC 11:31)
[000068684] . GPS NMEA source changed to port 1  h11:31:22 (UTC 11:31)
[000069193] .... RMC date field empty, skip sentence!
[000070219] .... RMC date field empty, skip sentence!
[000071202] .... RMC date field empty, skip sentence!
[000072212] .... RMC date field empty, skip sentence!
[000073212] .... RMC date field empty, skip sentence!
[000074228] .... RMC date field empty, skip sentence!
[000075203] [ASSERT FAILURE] in Common/Source/Geoid.cpp line 30
[000084095] ... GPS Port 1 no hearthbeats, but still gpsValid: forced invalid  h11:31:22 (UTC 11:31)
[000084106] ... GPS no active baro source, and still BaroAltitudeAvailable, forced off  h11:31:22 (UTC 11:31)
[000029132] ------------------------------------------------------------

pventafridda, 2016-02-20:
In another instance, after landing.. and again RMC date problem
[005269169] .... RMC date field empty, skip sentence!
[005270251] .... RMC date field empty, skip sentence!
[005271458] .... RMC date field empty, skip sentence!
[005272103] .... RMC date field empty, skip sentence!
[005273292] [ASSERT FAILURE] in Common/Source/Geoid.cpp line 30
[005281521] . Settings for 1258 airspaces saved to file <\Storage Card\LK8000\_Airspaces\AspConfig.LK>
[005282182] . CloseLKAirspace

brunotl, 2016-02-20:
Win CE5 or win CE6 ?

pventafridda, 2016-02-20:
Uhm  i have to check. I think it is a v2.
---
Sent from mobile

pventafridda, 2016-02-21:
Found the explanation , bug is fixed in the master branch.

closed 2016-02-21 by pventafridda