LK8000/LK8000 #825: LK8000 PC FullScreen distribution

SergioDaSilva82, 2016-02-26:
Is it possible to start releasing an exe for PC with FullScreen activated? Good for the Windows Tablets.

I would recommend a file name like: LK8000-PC-FS.exe

pventafridda, 2016-02-26:
Not in this beta!

From: Sérgio Da Silva 
Sent: Friday, February 26, 2016 1:28 AM
To: LK8000/LK8000 
Subject: [LK8000] LK8000 PC FullScreen distribution (#825)

Is it possible to start releasing an exe for PC with FullScreen activated? Good for the Windows Tablets.

I would recommend a file name like: LK8000-PC-FS.exe

SergioDaSilva82, 2016-02-26:
I didn't ask for this one, but if can be on the next one and in the V6 final version, I would be happy.

pventafridda, 2016-02-26:
Maybe it is better to have a command line parameter for full screen, not a standalone version.
We already allow to run LK8000.exe -x=1234 -y=543   for example.

From: Sérgio Da Silva 
Sent: Friday, February 26, 2016 1:51 AM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] LK8000 PC FullScreen distribution (#825)

I didn't ask for this one, but if can be on the next one and in the V6 final version, I would be happy.

SergioDaSilva82, 2016-02-26:
In my opinion just make more complicate to the non techsavvy people. It's easier just to click on the icon.

I think with some changes on the makefile and makeall would help for the next releases.

SergioDaSilva82, 2016-02-26:
https://github.com/LK8000/LK8000/commit/412be9d10a9d09427641c0731a7589510fc0b550

SergioDaSilva82, 2016-03-02:
make TARGET=PC FULLSCREEN=Y ?

AlphaLima, 2016-03-03:
Yes
FULLSCREEN=y 
works, but seems to have a problem in main map. All other pages and mutlimaps works fine in fullscreen mode

pventafridda, 2017-01-16:
It is now possible to simply drag and resize the PC window, so I think we can close this.

closed 2017-01-16 by pventafridda