LK8000/LK8000 #838: Dialog multiselection on close

pventafridda, 2016-03-08:
There is a possible problem with multiselection dialog. It is visible on PNA being slower than a PC. Long click on map for a multiselection, the list of items appears in dialog.
No selection, simple close. For a short while, the dialog is printed again with all items being the same: the first one in original list, and all them are selected in yellow.
Despite many attempts on PNA I cannot say it is creating any problem, so it is more likely a glitch.

brunotl, 2016-03-08:
Fixed

pventafridda, 2016-03-08:
yes I was closing it in fact thanks!

From: Bruno de Lacheisserie 
Sent: Tuesday, March 08, 2016 8:46 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Dialog multiselection on close (#838)

Fixed

closed 2016-03-08 by pventafridda