LK8000/LK8000 #841: Panning with OpenGL very slow

pventafridda, 2016-03-08:
Not a real problem for V6, but we are calling the UpToDate stuff at every move with OpenGL, hundreds of times, and this is slowing down panning to a border limit,

brunotl, 2016-03-08:
Interesting, how much time is spend in UpToDate() ?

pventafridda, 2016-03-08:
I don't know. It is called even 150 times in less than a second. But slow 
pan is not due to this. As you said we don't wait like on windows and we do 
refresh every move so it is normal to be slow.
Probably on fast cpu no problems and I close the issue.

---
Sent from mobile

brunotl, 2016-03-08:
150fps ? We can't says that slow ...

pventafridda, 2016-03-08:
All I can say is that on PC Linux at 2ghz it is slower than on a 500mhz pna
---
Sent from mobile

brunotl, 2016-03-09:
drag&move in pan mode profiling on LINUX i686 virtual machine

MapWindow::Render() call graph  (% of time & call count)

[image]

concerning UpToDate overhead :
   UpToDate() per call cycle estimation : 2 702
   DrawTerrain() per call cycle estimation : 3 663 376

so overhead is 0.073%

pventafridda, 2016-03-09:
Ok overhead of UpToDate is zero, I already acknowledged that.
I dont understand if you confirm the slow pan with OpenGL, although for a totally different reason.

From: Bruno de Lacheisserie 
Sent: Wednesday, March 09, 2016 3:57 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000] Panning with OpenGL slow (#841)

drag&move in pan mode profiling on LINUX i686 virtual machine

MapWindow::Render() call graph (% of time & call count)

concerning UpToDate overhead :
UpToDate() per call cycle estimation : 2 702
DrawTerrain() per call cycle estimation : 3 663 376

so overhead is 0.073%

closed 2016-03-17 by pventafridda