LK8000/LK8000 #851: Airspace alert missing and airspace dialog not responsive

pventafridda, 2016-03-21:
I started LK and changed in configuration the DEMO.txt airspace with the one attached, using demo maps and wp. 
[ITA_N_ASP2014.txt](https://github.com/LK8000/LK8000/files/183261/ITA_N_ASP2014.txt)
I have made replay of demo.igc and after takeoff you get an alert for airspace proximity, and immediately after an exit airspace closing the dialog, followed by another alert. This one could not be closed , keytouch not working (LINUX). After Close timeout, everything was ok. During that time no input working.

Trying to replicate, I once got takeoff and no airspace warning at all, just like the new configuration of airspace was not used.
I am doing more tests.

pventafridda, 2016-03-21:
To replicate missing airspace warning, here is the procedure using DEMO configuration.
Start with all demo things, but using ITA_N_ASP2014 airspace.
Enter configuration, and change airspace to DEMO.TXT. Exit configuration. 
Enter again configuration and change airspace again to ITA_N_ASP2014 .
Exit configuration, and replay log DEMO.IGC .
After takeoff at 12:35:53 and 12:35:59 you should get two alerts, but they will not work.
Exit and restart LK, replay demo.igc and alerts are working.

Suspect is 
4025730436] .... AIRSPACEFILECHANGED from configuration
[4025730437] . Settings for 15 airspaces saved to file </root/LK8000/_Airspaces/AspConfig.LK>
[4025730438] . CloseLKAirspace
[4025730438] . Reading airspace file
[4025730504] . Now we have 195 airspaces
[4025730504] . No airspace file 2

ASpConfig.LK which is related to old airspace file.

brunotl, 2016-03-22:
Regression commit : bb12a02a580d811902f092c7429bc94ccaf7443d

pventafridda, 2016-03-22:
The commit you want to reverse is needed for fast multimaps. I verified that the bug is existing also in v5. Maybe it is simpler to force a scanvisibility after reconfiguring airspaces, because it doesnt seem to happen out of that situation.
Concerning TopView ChangeDrawRect, we are inside Draw thread, where else should we change it?

brunotl, 2016-03-22:
ChangeDrawRect need to be done when we switch mapmode, not in draw thread.

currently, when we are in multimap, DrawRect is changed 2 time for each draw loop. so some part of draw is done with Screen bound and some others part are done with topview bound. that work bud generate overhead, and it's source of bug.

the only good way for fix that definitively is simple : separate screen layout calculation and drawing.

concerning commit : bb12a02, it's only workaround and duplicate code of SetTopologyBounds().

pventafridda, 2016-03-22:
I agree but the reason we have that piece of code is to reduce latency while changing multimaps. With a lot of waypoints, it is a problem. I dont remember anymore tests made, so all tests should be done again in multimaps with 3000 waypoints on a PNA. Otherwise, we risk a v6 very slow at changing pages. No other reasons..

brunotl, 2016-03-22:
this problem have 2 years.
adding workaround over workaround is wast of time, leave it as is, and take the time for fix the real problem.

pventafridda, 2016-03-22:
the problem is not a problem anymore, it was simply missing a check for ForceVisibilityScan . 
I am closing this issue and opening another for the key touch not responsive.

closed 2016-03-22 by pventafridda