LK8000/LK8000 #854: VARIO calculation broken

pventafridda, 2016-03-26:
The instant vario is totally broken. Compared to V5, replaying the same IGC, on v6 we see values jumping 28 m/s.

pventafridda, 2016-03-30:
in V6 the VARIO IS NOT CORRECT. 
All calculations using the vario are thus incorrect too.
(netto vario, MC etc.)

brunotl, 2016-03-30:
interesting, last change in Vario Calculation : 11/12/2012

brunotl, 2016-03-30:
what is your test case ?

pventafridda, 2016-03-30:
My test case is v5.0a.
We did not have any bug open on this, no problems reported, and consolidated working since 7 years. 
In 7 years we did hundreds of hours of tests. Several thousands of pilots have flown with it, and I have made over 800 hours in flight with absolutely no problems.
So if now the vario is not showing the same, while replaying an IGC file, then I assume it is not necessarily correct.
I hope to be wrong!

From: Bruno de Lacheisserie 
Sent: Wednesday, March 30, 2016 2:36 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] VARIO calculation broken (#854)

what is your test case ?

pventafridda, 2016-03-31:
Fixed by latest commit

closed 2016-03-31 by pventafridda