LK8000/LK8000 #855: STF speed to fly broken

pventafridda, 2016-03-26:
The STF is reporting crazy values for gliders.

pventafridda, 2016-03-30:
The commit b2846c5975df712ca976615e27621814d1de5c83  is changing the STF method. 
For gliders, and I believe also for PGs, the STF is often wrong,  showing a STF of minsink everytime.
Fast solution is revert the commit .

brunotl, 2016-03-30:
Polar DG 300 : 

MC 0    ->   99 km/h
MC 0.5 -> 110 km/h
MC 1    -> 121 km/h
MC 1.5 -> 131 km/h
MC 2    -> 140 km/h
MC 3    -> 156 km/h

nothing wrong.

pventafridda, 2016-03-30:
Comparing v5 on a glide, i read STF on v5 and 83-89 in v6.  Speed are quite always different.
In any case, it is wrong for gliders.

From: Bruno de Lacheisserie 
Sent: Wednesday, March 30, 2016 1:17 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] STF speed to fly broken (#855)

Polar DG 300 : 

MC 0 -> 99 km/h
MC 0.5 -> 110 km/h
MC 1 -> 121 km/h
MC 1.5 -> 131 km/h
MC 2 -> 140 km/h
MC 3 -> 156 km/h

nothing wrong.

brunotl, 2016-03-30:
v5 is not valid reference. only valid way for check is polar equation and mathematic ...

pventafridda, 2016-03-30:
Then the mathematics in v6 are wrong.
Because it is out of any question that the STF is wrong for gliders.

From: Bruno de Lacheisserie 
Sent: Wednesday, March 30, 2016 1:58 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] STF speed to fly broken (#855)

v5 is not valid reference. only valid way for check is polar equation and mathematic ...

pventafridda, 2016-03-30:
I am no more sure, I write to you.

closed 2016-03-30 by pventafridda