LK8000/LK8000 #859: Screen Mode not working in Linux, no rotation, no resize

pventafridda, 2016-03-27:
At least on Linux, SCREEN MODE is not resizing correctly.
Turn screen not working.
Screen size is not the desired one, and it is normally squared and bigger.

On PC it is working.

pventafridda, 2016-03-28:
Before september  it was working.

pventafridda, 2016-03-29:
Probably missing the AnnounceResize for OpenGL.

pventafridda, 2016-03-29:
Declassing to low priority, on Linux without OpenGL of course it does not happen. It is something to fix for version after 6.0

pventafridda, 2017-02-15:
It works for landscape to portrait, but not the opposite.

parapenT1sta, 2017-03-06:
Paolo,
You can rotate it properly but not at the first time.

For example:
Set 800x480 and then rotate screen, it goes "square"
but if you go again (without closing LK8000) and set 800x480 and rotate it again, already works fine.

Can you confirm it?

pventafridda, 2017-03-06:
Yes it is an old problem 

From: Sérgio Da Silva 
Sent: Monday, March 06, 2017 11:26 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda ; Author 
Subject: Re: [LK8000/LK8000] Screen Mode turn to portrait does not work on Linux (OPENGL) (#859)

Paolo,
You can rotate it properly but not at the first time.

For example:
Set 800x480 and then rotate screen, it goes "square"
but if you go again (without closing LK8000) and set 800x480 and rotate it again, already works fine.

Can you confirm it?