LK8000/LK8000 #861: Enhancement Infoboxes

de07y, 2016-03-28:
Hello,
I tried the "Pull request" menu but it looks like it is for code commit?

I just would want to submit an idea (enhancement, it is not an issue) that look important for me: (but perhaps is it very tricky to do it);
I am paragliding pilot.
When I flight, most of the time, when I know the area and when I am not using waypoint (so outside competitions): only 3 or 4 infoboxes are usefull and enough for me. That's one of the reason I tried LK8000 as it looks a little bit better using infoboxes.

On my old Garmin GPS 60CSX  I have a page with only 3 big infoboxes.

Imagine a page with only 3 or 4 Boxes (and no map) on the whole screen: Ground Speed, altitude, direction and a graphical Vario for example !

For me it is also something important for security: when you see big informations, your mind is not busy searching informations onto the screen...

That's the first Improvement I thought about and I would love to use.

The second one is : to have the possibility of using graphical informations into the infoboxes (!) like the Garmin 60CSX is doing it ( for the direction: an arrow inside the infobox is indicating it instead of a number ). This can be also very usefull for the live vario: seeing a graphical vario inside an info box
That's also can be very important for the security: if you see the ground speed at 2Km/h but with the arrow showing you are flying backward, then you know it.
With the number instead of the arrow, our mind will probably not calculate that the direction has been inverted... and it can be dangerous to miss that kind of information.

Hope this change will arrive one day, it would be so GREAT for me (and I think for other paragliding pilots).
Perhaps it is too tricky to change the code.
Don't hate me and just forgot my message if it is the case. :-)

Thanks for LK8000, it is really great.

Cheers,
Denis

pventafridda, 2016-03-28:
You can already remove overlays and leave only infoboxes. Depending on the size, 4 to 6, on the bottom.
No changes will be made on this matter, since they are already 200% configurable more than needed. If you want only overlays, reduce the size of bottom bar.
Graphics inside infoboxes are also ruled out: very nice in hangar, totally unusable in flight. Too small. 
I had a Garmin like yours too, years ago, before spending my time on LK. On LK you have the history of gotos working like on our garmins in fact. Copied from them.

de07y, 2016-03-28:
Hello, Thanks for your answer and for the informations. Sorry if I use the wrong words. I don't mean Map (map is another page) but info pages: I have a nice page but with more than 20 informations onto the page and I cannot reduce this number. For example: having only the bottom bar on the whole page would be perfect. There is a copyscreen of my Kobo screen I am using, but it is a lot of informations and  only with one line for example with huge text would be perfect.

But I undertand of course if it is too much modifications in the current code.
Thanks for all you are doing,
[image]

Cheers,
Denis

pventafridda, 2016-03-28:
As I told you, you can disable those pages and use only the bottom bar with even huge characters.
Or you can use overlays. All of them can be font-customized and content-customized. So I dont see the reason to redesign the infopages..
Plese check what I have just stated, and tell us if it is reasonable or if you cannot get to something reasonable.
best

From: de07y 
Sent: Monday, March 28, 2016 10:43 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] Enhancement Infoboxes (#861)

Hello, Thanks for your answer and for the informations. Sorry if I use the wrong words. I don't mean Map (map is another page) but info pages: I have a nice page but with more than 20 informations onto the page and I cannot reduce this number. For example: having only the bottom bar on the whole page would be perfect. There is a copyscreen of my Kobo screen I am using, but it is a lot of informations and only with one line for example with huge text would be perfect.

But I undertand of course if it is too much modifications in the current code.
Thanks for all you are doing,

Cheers,
Denis

de07y, 2016-03-28:
Yes, I tried a lot many days and again tonight trying the overlay and font size.
I would want to keep the map screen as second screen. As map is not very usefull for me excepted once arrived on the ground.. The copy screen is showing font size of the bottom bar at 6. If I put it at 10, then I have only the 3 first values (and all the other lines above the bottom bar are here and I didn't see any way to remove them: If I remove the whole page (and all the others, then I have only the map page. If I use the map page, I can also increase the size but it will not be like a page like this:
(Here with a quick look, we can see we are going back and up and all the screen display is used. (I put the graphical components  I thought about and can quickly speak to us more than numbers.)

[image]

Cheers,
denis

pventafridda, 2016-03-31:
There are plenty of instruments that can provide such display option. (Garmin is only one of them).
Maybe another version of LK , simplified, can be similar.
At the moment we have no plans to make a new version with this view.

closed 2016-03-31 by pventafridda

de07y, 2016-03-31:
Yes, a lot of instruments, and the reason is most probably because it is the best way to display the most importants informations. Thanks anyway for the time you spent looking at this proposal.  Cheers,
Denis