LK8000/LK8000 #867: OpenGeoid assertion failure line 30

pventafridda, 2016-04-01:
Report from a user who self compiled the version.

brunotl, 2016-04-01:
fixed long time ago : b6f74c5691368cbc4c3a0eb194d8b1ddc34d7fb4

pventafridda, 2016-04-01:
In fact this is the explanation I found on 20th february
ok, we apply geoidaltitudecorrection only if the gps is sending altitude without correction itself. If it is corrected, and notmally it is, then GGA fields 8 and 9 report the applied correction and we dont do anything else.
If it is not corrected because we dont have fields 8 and 9, then we apply correction on the geoid.
So the user had several RMC errors, gps fed from bt, probably a big mess incoming from the multiplexer in the V7 vario: at a certain point he get a GGA incoming from a second feed in the multiplexer, without geoid separation for some reason.
At that point, for the first time, parser is applying geoidcorrection .
In Geoid we had LookupGeoidSeparation, every time called by Parser.
      if (!egm96data || !OpenGeoid()) { 

.. and yes we have egm96data, perfectly loaded on startup. But we now go OpenGeoid, mistake.
OpenGeoid says what the hell, we already have it loaded.. and LKASSERT pops up. Bang.

He said he is using latest version, and it is not possible unless he did not a clean build..

From: Bruno de Lacheisserie 
Sent: Friday, April 01, 2016 3:44 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] OpenGeoid assertion failure line 30 (#867)

fixed long time ago : b6f74c5

brunotl, 2016-04-01:
2 fact : 
  - OpenGeoid() is called only once: lk8000.cpp line 425.
  - no assert line 30, removed 20/02/2016 now assert is line 29.

pventafridda, 2016-04-01:
Right!

From: Bruno de Lacheisserie 
Sent: Friday, April 01, 2016 3:58 PM
To: LK8000/LK8000 
Cc: Paolo Ventafridda 
Subject: Re: [LK8000/LK8000] OpenGeoid assertion failure line 30 (#867)

2 fact : 

  a.. OpenGeoid() is called only once: lk8000.cpp line 425. 
  b.. no assert line 30, removed 20/02/2016 now assert is line 29.

closed 2016-04-01 by pventafridda